keyboardio/Kaleidoscope #3: Error building mouse_movement.o

nuclearsandwich, 2015-07-03:
```
[0] $ make
/usr/bin/avr-g++ -c -g -w -Wa,-adhlns=build/mouse_movement.lst -DVERSION=\"e220-dirty\" -DUSB_VID=0x1209 -DUSB_PID=0x2301 '-DUSB_PRODUCT="Keyboardio model01"'  -DARDUINO_ARCH_AVR -DUSING_MAKEFILE=1 -Wall -Wextra -Wformat=2 -Wuninitialized -Wshadow -Wconversion  -Os -freorder-blocks -fno-inline-small-functions -fno-exceptions -ffunction-sections -fdata-sections -MMD -DF_CPU=16000000L -DARDUINO=165   -mmcu=atmega32u4 -I/usr/share/arduino/hardware/arduino/avr/libraries/EEPROM -I/usr/share/arduino/hardware/arduino/avr/libraries/Wire -I/usr/share/arduino/hardware/arduino/avr/libraries/SPI -I/usr/share/arduino/hardware/arduino/avr/libraries/SoftwareSerial -I/usr/share/arduino/hardware/arduino/avr/libraries/EEPROM -I/usr/share/arduino/hardware/arduino/avr/libraries/Wire -I/usr/share/arduino/hardware/arduino/avr/libraries/EEPROM -I/usr/share/arduino/hardware/arduino/avr/libraries/Wire -I/usr/share/arduino/hardware/arduino/avr/libraries/Wire/utility -I/usr/share/arduino/hardware/arduino/avr/libraries/Wire/utility -I/usr/share/arduino/hardware/arduino/avr/libraries/Wire/utility -Ihardware/keyboardio/avr/variants/model01  -Ihardware/keyboardio/avr/cores/keyboardio -MD -MT 'build/libsketch.a/(mouse_movement.o)' -MF build/libsketch.a/._mouse_movement.dep mouse_movement.cpp -o build/mouse_movement.o
mouse_movement.cpp:157:1: fatal error: opening dependency file build/libsketch.a/._mouse_movement.dep: No such file or directory
 }
 ^
compilation terminated.
_Makefile.Master:661: recipe for target 'build/libsketch.a/(mouse_movement.o)' failed
make: *** [build/libsketch.a/(mouse_movement.o)] Error 1
```

I'm not sure why this is specifically missing or why there are parens around the file name.

obra, 2015-07-04:
huh. What linux? I'll set up a repro environment.

Oh. Actually, try building using the arduino ide?

nuclearsandwich, 2015-07-04:
```
[0] $ uname -a
Linux totoro 4.0.6-1-ARCH #1 SMP PREEMPT Tue Jun 23 14:25:08 CEST 2015 x86_64 GNU/Linux
[0] $ avr-gcc -v
Using built-in specs.
COLLECT_GCC=avr-gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/avr/4.9.2/lto-wrapper
Target: avr
Configured with: /build/avr-gcc/src/gcc-4.9-20141224/configure --disable-install-libiberty --disable-libssp --disable-libstdcxx-pch --disable-libunwind-exceptions --disable-linker-build-id --disable-nls --disable-werror --enable-__cxa_atexit --enable-checking=release --enable-clocale=gnu --enable-cloog-backend=isl --enable-gnu-unique-object --enable-gold --enable-languages=c,c++ --enable-ld=default --enable-lto --enable-plugin --enable-shared --infodir=/usr/share/info --libdir=/usr/lib --libexecdir=/usr/lib --mandir=/usr/share/man --prefix=/usr --target=avr --with-as=/usr/bin/avr-as --with-gnu-as --with-gnu-ld --with-ld=/usr/bin/avr-ld --with-plugin-ld=ld.gold --with-system-zlib
Thread model: single
gcc version 4.9.2 20141224 (prerelease) (GCC)
```

Building with Arduino IDE gave me

```
Arduino: 1.6.5 (Linux), Board: "Arduino Uno"

mouse_movement.cpp: In function 'void _warp_jump(long int, long int, long int, long int)':
mouse_movement.cpp:13: error: 'Mouse' only supported on the Arduino Leonardo                 Mouse.moveAbsolute(x_center,y_center);
                 ^
mouse_movement.cpp: In function 'void handle_mouse_movement(char, char)':
mouse_movement.cpp:132: error: 'Mouse' only supported on the Arduino Leonardo     Mouse.move(moveX, moveY, 0);
     ^
'Mouse' only supported on the Arduino Leonardo

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.
```

Wonder why it's selecting the wrong arduino? Should I be specifying a bundled arduino home rather than the system arduino packages?

obra, 2015-07-04:
Oh. If you're in the Arduino IDE, you need to select the Keyboardio Model 01 board.

https://github.com/obra/KeyboardioFirmware/wiki/Arduino-Setup-Linux
followed by 
https://github.com/obra/KeyboardioFirmware/wiki/Flashing-the-firmware

My suspicion is that you missed step 3 in Flashing-the-firmware.

nuclearsandwich, 2015-07-04:
Oh word. Didn't even think to look in the wiki for info. My bad. Expect a PR adding a readme that includes links to the wiki :wink:

nuclearsandwich, 2015-07-04:
Selecting the correct hardware in the IDE definitely set it up. The Arduino IDE is not hidpi friendly apparently. 

[image]

Looking into the [ino utility](http://inotool.org/). If I find it works well I'll add to the wiki.

closed 2015-07-04 by nuclearsandwich

obra, 2015-07-04:
On Sat, Jul 04, 2015 at 07:33:24AM -0700, Steven! Ragnarök wrote:
> Selecting the correct hardware in the IDE definitely set it up. The Arduino IDE is not hidpi friendly apparently. 

Oy. That ought to get fixed.

> Looking into the [ino utility](http://inotool.org/). If I find it works well I'll add to the wiki.

Thanks. We also need to replace the old makefile build system with the Arduino 1.6 equivalent. I'm told that somewhere deep inside 1.6 is a commandline build system.

-jesse