keyboardio/Kaleidoscope #4: NOTICE and LICENSE mismatch (simple fix)

silverhook, 2015-10-31:
I noticed that the `LICENSE` file now says GPLv2, which seems fine.

But the `NOTICE` file still mentions that the Keyboard.IO driver is licensed under the Apache 2.0.

I’d suggest you update the NOTICE file to point to the right license.

Since `NOTICE` file is not needed under GPL (but is needed if you contribute to an Apache Foundation project), it might even make sense to change the `NOTICE` file to `README` or `README.markdown`. Feel free to keep the filename as it is, if that suits your style better. Not a biggie there :smile_cat:

obra, 2015-11-03:
>    [1]:smile_cat:
>    Reverse link: [3]unknown
> References
>    Visible links

Thanks. (Our ideal is to license under apache 2.0, but the LED driver
we're currently using is GPL-only.)

I've removed the conflicting statement from NOTICE, but left the file, 
as it describes the provenance of the arduino core. Other third-party
code should be noted there as well

silverhook, 2015-11-08:
OK, I understand your idea.

The issue is that now the `NOTICE` file looks as if you apply full copyright on your driver, which could be interpreted as conflicting with the license.

We can look into this later on when we both have time, especially since, as I understand, you’re rewriting the firmware.

I haven’t checked it in great detail, but if you’re referring to the code in `hardware/keyboardio/avr/` – most of that stuff looks under **LGPL-2.1+** (GNU Lesser General Public License, version 2.1 or later). That license does not extend to the whole project, since it limits itself to the “library”.

I can look into the licensing options, if that’d help you guys. If so, it’d help to know what the reasons are for your choice of Apache 2.0 and what you’d like to achieve with your license.

obra, 2017-03-27:
The NOTICE file has been removed

closed 2017-03-27 by obra