keyboardio/Kaleidoscope #13: Support keys with different behavior when held or tapped (e.g. A Ctrl/Esc key)

obra, 2016-07-01:


algernon, 2016-08-02:
I'd like to chime in here, and note that I'm working on a featured called 'tap-dance' in QMK, upon which many things can be built, such as double-action keys, double-tap keys, one-shot stuff and more. Once the core feature is ironed out in QMK, I'd be happy to port it - and the one-shot stuff, double-action and double-tap things - to keyboardio too, if you're interested in them.

obra, 2016-08-02:
<3 If we don't get to it first, we'd love that.

algernon, 2016-11-22:
FWIW, I have a variant of this [here](https://github.com/algernon/Akela/blob/master/lib/Akela-SpaceCadet/src/Akela/SpaceCadet.cpp), and the `Ctrl/Esc` (modifier/key, in general) will be [coming soon](https://github.com/algernon/Akela/issues/56). So even if not in the stock firmware, the feature will be available as a plugin by the time the keyboards ship :)

algernon, 2017-01-24:
[Akela-DualUse](https://github.com/keyboardio/Akela-DualUse) does just this, for both modifiers and layer keys, and is included in `keyboardio-libraries`, so I suppose this can be resolved? :)

algernon, 2017-03-28:
[DualUse](https://github.com/keyboardio/Kaleidoscope-DualUse) is exactly this.

closed 2017-03-28 by algernon