keyboardio/Kaleidoscope #14: Negotiate power management with the host (Only pull 100ma on iOS)

obra, 2016-07-01:


obra, 2017-11-26:
As one reference, search for 'power_neg' in https://github.com/kiibohd/controller/blob/0203b275e9bf16b5743d417e25295b366f9ff95a/Output/pjrcUSB/arm/usb_dev.c

obra, 2017-11-26:
https://github.com/kiibohd/controller/blob/cc42d04138b1ea8791da91200d23f5388dfb5b21/Output/pjrcUSB/capabilities.kll#L61 describes possible issues with this scheme.

algernon, 2018-10-18:
This is still an issue, 81179aa is another repo's thing.

harritaylor, 2021-02-12:
I believe I may be facing this issue at the moment with my Atreus. My iPad doesn't complain when the keyboard is connected and in use, but after ~5 minutes of no key inputs, my iPad restarts. After talking to Apple support they suggested it may be the power draw from the keyboard. If you could point me in the right direction I'd be happy to take up work on this issue and contribute if it hasn't been resolved. Thanks!

obra, 2021-02-12:
Hi Hari,

Can you tell us which kind of iPad you're using and whether this happens
only with the Atreus or also with other keyboards? The Atreus doesn't draw
much power at all and based on my discussions with engineers at Apple, it's
my understanding that iOS will cut off devices that are drawing power, but
shouldn't itself then crash or reset.

harritaylor, 2021-02-13:
Hi @obra, thanks for getting back to me so quickly. It’s the 2018 regular iPad 6th generation. I’ve tried to test as best I can: so far I cannot replicate the issue using a Leopold FC660m. I reset my iPad to factory settings and kept the Atreus plugged in overnight and unfortunately it triggered the crash and restart again.
I guess I will try and get the iPad replaced in that case; I’ve not read anything about other people having this issue, and it seems that it’s more an Apple problem.