keyboardio/Kaleidoscope #9: The user should be able to drive the LEDs over a serial connection from the host

obra, 2016-07-01:


algernon, 2016-12-01:
Quick idea:

Have a small library that insalls a `loop` hook, and checks if there is data available on `Serial`. If so, it reads a single byte, the command ID, and checks if there is any handler registered for that command. If not, it sends back a `NACK`, so the host program won't send more data. If there is, it sends an `ACK`, and passes handling to the handler.

One such handler would be the LED updater, which would have the following sub-commands (they'd be IDs, not strings, I'm just listing their names here for clarity):

- `NextEffect`: Like the key of similar name. No further arguments.
- `SelectEffect`: A single byte argument, selects the effect with the supplied index.
- `SetColor`: Takes two coordinate bytes, and a three-byte cRGB. Sets the LED at row,col to the given color.
- `SetAllColors`: Sets all leds to the specified color.
- `SetColormap`: First argument is a one-byte boolean: with or without transparent colors? If with, takes three extra bytes: the cRGB values that will be treated as transparent. After these, takes 64 cRGB values, and applies them.
- `GetColors`: Returns the LED colors, in the key layout order.

At the end, all commands close the communication with another `ACK`.

What do you think? This sounds pretty easy to do, and the host program wouldn't be all that hard, either. I can likely write one in either of bash, python or Perl, whichever you prefer.

algernon, 2016-12-01:
Also, if I'll go ahead with this, do you want it in core, like the macro and mousekey plugins, or shall I put it in Akela?

obra, 2016-12-01:
>    Reverse link: [3]unknown
> References
>    Visible links

I do want something like this in the core. Ideally, it would be able to
be used for other things like:

* Debugging output
* Layout updates
* Dumping the current layout
* Device info
* Playing a nice game of colossal cave
* Taking input from the host about the current application's name so it
  can offer app-specific macros
* Being told by the host what OS it's running, so we can do unicode
  input well.

I'd been envisioning a textual protocol that we could repurpose as the
on-device programming system. You'd open up a text editor, type the
magic startup phrase and be presented with a command prompt.

The keyboard would take typed input and respond by typing its output.

I mocked this up in some early firmware build a couple years ago. The
code was awful, but the idea felt neat.

obra, 2016-12-01:
Core please.

algernon, 2016-12-01:
Hrm, textual has the advantage of being friendly, and, looking at the Arduino libraries, we have int parsers anyway, so... it's not that much more code than binary protocols. And speed doesn't matter, either.

I'll cook something up!

cbbrowne, 2016-12-01:
Think... IETF protocol, ala SMTP (RFC 5321).
With SMTP, you are talking to an email server, so you say..
EHLO, to greet the server.
MAIL FROM, to say who's to be sending mail.
RCP TO, indicating recipient.
DATA, indicating the content.

For a keyboard, there are several areas of protocol:
- Greeting it, of course.  HELLO.
- Telling it about key mappings.  MAP a key TO a behaviour, usually to emit an output.
- Telling it what to do to Das Blinkenlightzen.  Perhaps some of the behaviours include "blinks".
- I'll bet there are some asynchronous events to be managed and triggered.
- Probably some timers will initiate activities.  Such as "keep lights quiet after 10 minutes of silence".
- Dump out all of the above state.
- Here's a blob; do CRC check on it, and if it's all good, then write it as a firmware update.
- Tell me what blob you have in your firmware.
- Tell me about yourself.  Version numbers, numbers of keys, and how they are addressed.  How much memory is there, how much is free.

algernon, 2016-12-02:
I'd like to make this extensible, so the textual protocol should be done in such a way that commands, and all their arguments and data, can be discarded without implementing said command. In this case, the keyboard would send back an error.

Here's one idea (`<` is the host, `>` is the keyboard's response):

```
< hello
> ERROR 1
> Incomplete handshake. How rude.

< hello keyboard, my old friend
> OK 1
> you've come to talk with me again

< commands
> OK 1
> hello commands settings host LED

< settings
> OK 2
> settings.defaultLayer = 0
> settings.useEEPROM = 0

< settings.defaultLayer = 1
> OK 1
> settings.defaultLayer = 1

< host.currentProgram = Emacs
> OK 1
> host.currentProgram = Emacs

< host.lalala
> ERROR 1
> host.lalala is not a valid command.

< LED.colormap = #000000 #ffffff ....
> OK 0
```

So basically, every command would be a single line, which makes it easy to ignore them. To save memory, the command name itself would be limited to, say 32 bytes or something. The firmware would read into a buffer until it finds a newline, space, or `=`. Everything before that is the command, and handlers can be registered to handle the rest. The rest would not be buffered, as that'd eat lots of memory. It will be up to the handler to read the rest, in whatever way it prefers. For unknown commands, we'd read until the end of the line, and report an error. All responses are either `OK` or `ERROR`, followed by the number of lines output.

There is one exception to the every command on one-line rule: if the line starts with whitespace, it is a continuation. This makes it nicer to send larger amount of data in a human-friendlier format, but still allows the firmware to ignore the data if there is no handler.

Thus, the LED colormap example could look like this:

```
< LED.colormap = #000000 ...
                 #ffffff ....
                 ....
> OK 0
```

This complicates reading the data a little bit, but not by much.

Oh, and the `hello` command would be needed to enable the rest of the commands. Without a proper handshake, all other commands would return the same error message. Of course, the required parameters, the greeting and the default error message would be customizable, too.

algernon, 2016-12-02:
I started to sketch out an implementation on my [WIP/SerialCommander](https://github.com/algernon/KeyboardioFirmware/commits/WIP/SerialCommander) branch. It's at a very early stage, but the size increase is substantial (almost +4k code, +800byte data), so I may have to revisit some of the decisions made.

Making `LED` and `LED.colormap` different commands sounded simple, but that requires far too much memory.

Back to the drawing table, somewhat. I still think that the protocol is reasonable, but the implementation I currently have, is not good. It's enough for a PoC, to see if the idea works at all, but not suitable for anything more than that yet.

algernon, 2017-01-03:
I've been thinking about this for a while now, and I'm becoming more and more convinced that the textual protocol has more problems than the good things it buys us.

1) It is non-trivial to come up with an extensible, text-based protocol that we can parse with very limited amount of memory needed.
2) Even if we did come up with such a protocol, parsing it still takes considerably more memory than a binary protocol. And memory is something we're very short on.
3) The end-user still needs at least a small tool to talk to the keyboard, even over a textual protocol. We could build the text->binary protocol translation into the same app to begin with.

So I'm leaning towards trying a binary protocol, with the following properties:

- Extensible
- Discovereable
- Efficiently parsable

We'd start with having a `command` byte, of which some are reserved, and we'd have an enum that ends with, say, `KEYBOARDIO_PROTOCOL_SAFE` or something, so plugins can start their numbering from there. The command byte would be followed by the full length of the rest of the data, on 16 bits. This is most useful for cases where we don't implement the command, so we can read till the end and handle things gracefully.

What is in the data part, is up for each command. It can be another command byte, or something else.

There would be a `describe` command, that would dump the known commands, their parameters, and so on. Plugins would be required to fill this in, so their commands can be dumped too (for discovery purposes).

-----

We could use a tree-like structure to make commands extensible, and discoverable. As an example, to turn layer 2 on, we could send the following bytes:

- `0x04`: ID of the `layer` item, which is a container.
- `0x02`: length of the payload, past this byte
- `0x01`: the `on` command's ID in the `layer` container.
- `0x02`: the layer number.

Internally, we'd look at our `items` array, and find the 4th element, see that it is a container. We store the length of the payload, and see the `on` command: we look that up within an array pointed to by the `layer` item, see that it is a command, and call the callback it points to with our Serial reader set to point to the next byte, and the remaining length of the payload. The callback would be responsible for parsing the rest, whatever way it sees fit.

In other terms:

```c++
typedef enum {
  CONTAINER,
  COMMAND
} item_type_t;

typedef void (*command_t) (uint8_t commandID, uint16_t payloadLength);

typedef struct _control_t {
  item_type_t type;
  char *description; // this should be a PROGMEM-stored string
  union {
    struct {
      uint8_t itemCount;
      _control_t *items;
    } container;
    command_t callback;
  }:
} control_t;

extern control_t controlStuff[] = { .... };
```

The `describe` command can just walk this structure recursively, and print stuff, indenting a bit more at each level.

Oh, and this also makes it possible to have different arities: a `describe` with a zero remaining payload means *describe everything*. Each additional byte means that it will try to look for that particular item in the tree. For example `describe 1 2` will look at the first item in `controlStuff`, and assuming it is a container, look at the second element in its `items` property, and dump the `description` of that.

Parsing this is reasonably simple. Extending too. And thanks to the discoverability, plugins and agent-like tools do not need to care about what the ID of a particular command is, they can dig that out of the description. We just need to follow a convention to make that easier, and provide macros or other helpers for plugin authors.

algernon, 2017-02-25:
I have made a [huge breakthrough](https://github.com/keyboardio/Kaleidoscope-Focus): a plugin, that can listen to serial commands, and reply back. It is extensible, without any arrays pre-allocated (so you can go and have as many handlers as you can fit into your firmware), and built-in support for self-documentation, and uses a text-based protocol.

All in <500 bytes of program code, and 60 bytes of data. Each new command adds a bit of code (about 36 bytes + the documentation, on top of the handler code itself), and  about 14 bytes of data.

There is very little error handling, so great care must be taken to send well-formed data to the keyboard. But these numbers are, I think, pretty darn good.

algernon, 2017-03-28:
Fixed with the combination of [Focus](https://github.com/keyboardio/Kaleidoscope-Focus) and [LEDControl](https://github.com/keyboardio/Kaledioscope-LEDControl).

closed 2017-03-28 by algernon

algernon, 2018-10-16:
Due to the plugin being merged into the [Kaleidoscope](https://github.com/keyboardio/Kaleidoscope) monorepo, issues are being migrated over there. This issue is going to be tracked further as keyboardio/Kaleidoscope#374.