keyboardio/Kaleidoscope #21: Librarification, and using external libraries

algernon, 2016-11-10:
So the other day I prepared a PoC, see #20, and that got me thinking about how to make it easier to use anything I write for Akela, with KeyboardioFirmware. Then it dawned on me, that this may be actually very simple!

There are two parts to this:

1. Making it possible for Akela libraries to use KeyboardioFirmware as if it was an Akela.
2. Making it possible for KeyboardioFirmware to use Akela libraries.

For the first, if I can use KeyboardioFirmware as a library, then I can write a wrapper that makes it look like Akela, and present the same APIs, and as such, Akela libraries will just work fine.

For the second, a user needs to be able to hook into the main event loop, and into the key event processing flow, and insert their own code, that calls out to Akela libraries. The former is already possbile, with the main `loop` function in the Sketch: we only need to call Akela's keyboard loop function instead of `Keyboard.loop()`. The latter, hooking into key processing is a bit trickier, but not by much: create a `bool handle_user_key_event(byte row, byte col, uint8_t currentState, uint8_t previousState)` function. This would return `true` if the event was processed by this function, and should not be processed further; and false otherwise. In `handle_key_event`, we'd call this function first, and return early if it returns true. This allows the user to override what happens during key handling, and insert new behaviour, such as using the Akela libraries.

By default, this `handle_user_key_event` function would simply return `false`. We'd also tell the linker to mark it `weak`, so any other definition of the same function would override it. This makes it possible to use it from a user Sketch, and hook into the event processing.

All in all, this would be just a few lines of code needed in KeyboardioFirmware, and would add minimal to the code size. Yet, would make it possible to hook into event processing, and make the firmware far more extendable than it currently is, without over-complicating things. May not be the cleanest, nicest way, but it is effective.

What do you think? I can prepare a PoC to show off what I was thinking of. A simple one, that just extends the firmware, without using other libraries, to make it a simple one.

algernon, 2016-11-10:
Created a simple PoC, to show what I was thinking of.

obra, 2016-11-10:
Seems reasonable. I'd be willing to take that patch.

algernon, 2016-11-10:
I'll turn this into a PR in a bit then.