keyboardio/Kaleidoscope #38: Left hand modifiers + right hand keys question

algernon, 2016-11-15:
I got the following question on the wiki of my fork, from @NicholasSterling, copying it here to make it easier to answer:

------

1) Can left-hand modifiers affect right-hand keys (and vice versa)?

In `Model01.cpp`, `act_on_matrix_scan()` makes two calls to `handle_key_event` -- the first takes the left-hand state, and the second takes the right-hand state. Does that mean that we could not have `Ctrl` and `Alt` under the left thumb and `Shift` under the right thumb, and use them to produce, say `Ctrl-Alt-Shift-L`?

algernon, 2016-11-15:
I'll try to answer this to the best of my knowledge, but keep in mind that I'm not deeply familiar with neither the firmware, nor the hardware (yet).

So, the way key events are handled, is that each scan cycle, the events are collected, and `handle_key_event` goes through them. When it *presses* a key, that is not immediately sent to the host computer. Instead, one full report is sent after processing all keys, combining all states from both halves. This means that you can press `Ctrl` and `Alt` with one thumb, and `Shift` and `L` with the other, and the host will receive that as `Ctrl+Alt+Shift+L`, as one would expect.

To put it another way: `handle_key_event` just collects the actions that need to be done, it assembles a report, which is sent to the host at the end of the scan cycle, all in one. So the events of the two halves are combined into one single report, and the events do not happen immediately, but in bulk.

NicholasSterling, 2016-11-15:
Thanks, Gergely — that sounds great.

Nicholas

algernon, 2016-11-16:
Glad I could be of assistance! Closing the issue, as the question appears to be answered sufficiently :)

closed 2016-11-16 by algernon