keyboardio/Kaleidoscope #67: Add `Miami` theme

menelaos, 2016-12-08:
Would be nice to have a LED color mode corresponding to the popular `Miami` keycaps.

algernon, 2016-12-08:
I'm happy to add this to [Akela](https://github.com/algernon/Akela), as LED effects are something that need not be in core. Do you have a link to a Miami photo that would be a good basis for the effect?

menelaos, 2016-12-08:
Cool, thanks! [This](http://i.imgur.com/lJmhfbM.png) image might give you a good indication. An image search for "Miami keycaps" will also yield a lot of results.

obra, 2016-12-08:
I would love to have LED affect plug-ins for things like this. This seems like a good example of a simple custom LED effect. It might be the ideal first one to put in an example repository to clone and fork

algernon, 2016-12-08:
Yessir, on it! I'll name it `Keyboardio-LEDEffects`. I have a couple of neat ideas how to make it friendly.

obra, 2016-12-08:
I was thinking of a plug-in for a single effect. So that when anybody wants to create a new effect, they can just create a new tiny plug-in copying from this

algernon, 2016-12-08:
Aha, I see. Not sure that's worth it, because a LED effect is almost trivial to implement: a class that subclasses `LEDEffect`, an `update` method, and a constructor that registers it. Copying from `LED-Off`, or any of the current effects in core is about as simple as cloning a repo with dummy functions... ok, perhaps cloning is marginally easier, but I'd think that having a library with a variety of effects would be more useful, because then people who want to write their own, have more examples to look at, or borrow from.

algernon, 2017-01-24:
With the recent change of including Akela plugins in arduino-boards, this is pretty much done, I believe, as the [Akela-LEDEffects](https://github.com/keyboardio/Akela-LEDEffects) plugin has a [Miami effect](https://github.com/keyboardio/Akela-LEDEffects#miamieffect).

algernon, 2017-03-28:
For reference, the plugin is now called [LEDEffects](https://github.com/keyboardio/Kaleidoscope-LEDEffects), and the theme's [here](https://github.com/keyboardio/Kaleidoscope-LEDEffects#miamieffect).

closed 2017-03-28 by algernon