keyboardio/Kaleidoscope #120: ConsumerControl does not repeat

algernon, 2017-03-13:
At the moment, `ConsumerControl` keys are not repeating, because we use `ConsumerControl.write()`, and only when the key toggles on. This means that if I want to fiddle with volume control (iirc that is part of `ConsumerControl`, but `SystemControl` suffers from the same issue), I need to repeatedly tap a key, instead of being able to hold it.

I think we should be using `.press()`, and `.releaseAll()` in the loop, like we do for `Keyboard`, but I have to test if this works as one would expect.

obra, 2017-03-13:
Note that the switch to write is a recent change. Until recently, the way we were doing it would cause even the briefest tap to send so many events to the host that it would overwhelm it with repeats. The specific symptom is that tapping the volume up key briefly on a mac would push it up to max volume and send so many extra events, volume couldn't be dropped back down.

This may be something where we want to handle CC events marked "OSC" (one shot control) differently than others.

algernon, 2017-03-13:
Mhm, I suspected that may be behind the behaviour. I'll see what / how other keyboards do it, and experiment a bit.

algernon, 2017-03-13:
Ah, there we go. Problem with the previous implementation, which only used `.press`, was that it never released the key. Merely changing `.write` to `.press` does, indeed, exhibit the problem you described, and the key remains "stuck" forever. If I add an explicit `.release` when the key toggles off, it all works fine.

PR #121 (2017-03-15): Support for holding Consumer/SystemControl keys
Instead of calling the `.write` method of `ConsumerControl`/`SystemControl`, which registers the key, sends a report, then unregisters, and reports again, just `.press` it when the key is pressed, as we did before.

However, since `.press` always sends a report, and so does `.releaseAll`, we can't have it the same way we do for `Keyboard`. We need to explicitly release the consumer/system key, when the triggering key is released too. Not doing so makes the key stuck, as we never release it, and that will upset the operating system very much.

Wit [... cut]
commit ff4dfad1 (2017-03-15): Support for holding Consumer/SystemControl keys

Instead of calling the `.write` method of `ConsumerControl`/`SystemControl`,
which registers the key, sends a report, then unregisters, and reports again,
just `.press` it when the key is pressed, as we did before.

However, since `.press` always sends a report, and so does `.releaseAll`, we
can't have it the same way we do for `Keyboard`. We need to explicitly release
the consumer/system key, when the triggering key is released too. Not doing so
makes the key stuck, as we never release it, and that will upset the operating
system very much.

Wi [... cut]
closed 2017-03-15 by obra