keyboardio/Kaleidoscope #126: Consider some sort of "tap" helper

obra, 2017-03-27:
handle_keyswitch_event (Key_Backspace, UNKNOWN_KEYSWITCH_LOCATION, IS_PRESSED | INJECTED);
          Keyboard.sendReport ();
          handle_keyswitch_event (Key_Backspace, UNKNOWN_KEYSWITCH_LOCATION, WAS_PRESSED | INJECTED);
          Keyboard.sendReport ();

is a pattern we repeat more than a few times. Perhaps we should have a helper for it

algernon, 2017-03-28:
What would be neat, imo, is something like this:

```c++
void handle_keyswitch_events (Key mappedKey, byte row, byte col, uint8_t keyStates) {
    if (keyStates & IS_PRESSED) {
      handle_keyswitch_event (mappedKey, row, col, IS_PRESSED | INJECTED);
      Keyboard.sendReport ();
    }
    if (keyStates & WAS_PRESSED) {
      handle_keyswitch_event (mappedKey, row, col, WAS_PRESSED | INJECTED);
      Keyboard.sendReport ();
    }
    if (!(keyStates & IS_PRESSED) && !(keyStates & WAS_PRESSED)
      handle_keyswitch_event (mappedKey, row, col, keyStates | INJECTED);
}
```

Kind of... this would work for the above case, and for those cases where we only want to handle a press or a release, with a report sent immediately after (plenty of those scattered around in various plugins). Would need a bit more work to correctly set the `keyState`, so that it preserves what needs to be preserved... perhaps `keyState & ~WAS_PRESSED` in the `keyState & IS_PRESSED`  state... the goal is to handle each state separately. The disadvantage is that this does not allow a "held" state, but iirc, that's rarely used by existing plugins - but I'll have a closer look.

What would be even more useful, is:

```c++
void handle_keyswitch_event (Key mappedKey, uint8_t keyState) {
  handle_keyswitch_event (mappedKey, UNKNOWN_SWITCH_LOCATION, keyState);
}
```

This would make it more convenient to trigger key handling from unknown locations, by hiding the `UNKNOWN_SWITCH_LOCATION` macro.

gedankenexperimenter, 2019-05-06:
I have the feeling that this wouldn't be as generally useful as it seems. Macros has the most apparent need for it, but already has its own wrapper. Maybe it's just time to close this one?

algernon, 2019-05-10:
I agree, closing accordingly. Please reopen @obra if you disagree.

closed 2019-05-10 by algernon