keyboardio/Kaleidoscope #127: Feature request: Quasi (aka Leep) key

Pitometsu, 2017-03-31:
Universal modifier key.
***
Main idea: "press Quasi", "press <trigger key> (e.g. Ctrl)", "release <trigger key>", "press <target key>" (with holden "Quasi") => acts like "modifier key" (from "trigger to modifier map" according to pressed <trigger key>) plus "<target key>".

It may be convenient when you need to hold few modifiers.
Example:
- hold "Quasi"
- press and release "Alt"
- press and release "Ctrl"
- press "D"
=> "Alt+Ctrl+D" as result.

we can customize "trigger to modifier map" to press e.g. "C" key for "Ctrl" modifier.

[Here](http://ibnteo.klava.org/keyboard/quasi) original idea author's blog. I'm not the author. Initially idea based on Jef Raskin's principles from "The Humane Interface".

Pitometsu, 2017-03-31:
Also it can help with limitation when we forced to press few modifier keys by single thumb.

algernon, 2017-03-31:
This sounds very much like [One-Shot modifiers](https://github.com/keyboardio/Kaleidoscope-OneShot), except with one-shots, you do not need the Quasi-key, you just tap `Ctrl`, `Alt`, and `d` to get `Alt+Ctrl+D`.

Though, reading the original article, it is closer to [Leader](https://github.com/keyboardio/Kaleidoscope-Leader), the main difference is that for Leader, you don't have to hold the Leader key, either.

Would Leader be a close enough approximation?

Pitometsu, 2017-03-31:
Similar, but not the same. According to Jef Raskin's principles, modes are evil, but holding key let our mind keep it explicit (that's where Quasi name from: quasi-modes. Modal editing without modes). With leader key it's pretty simple to forgot that you pressed prefix (any attention refocus). With key holding it's enough just release key - and mode will be forgotten. See "The Humane Interface" for details.

algernon, 2017-03-31:
Mhm. I'll see if I can teach `Leader` to work its magic while held, too... that should do the trick, I hope.

Pitometsu, 2020-05-06:
It supposed to work like:
- keep modifiers while `Quasi` pressed
```md
- hold "Quasi"
- press and release "Alt"
- press and release "Ctrl"
- press "D"
=> "Alt+Ctrl+D" as a result.
```
- forgot modifiers when `Quasi` released
```md
- hold "Quasi"
- press and release "Alt"
- press and release "Ctrl"
- release "Quasi"
- press "D"
=> "D" as a result.
```
- keep modigiers while switching the hand
```md
- hold "Quasi"
- press and release "Alt"
- press "Quasi" (e.g. other hand)
- press and release "Ctrl"
- release "Quasi"
- press "D"
- release "Quasi"
- press "D"
=> "Alt+Ctrl+D", "D" as a result.
```
- yet forgot mode on release
```md
- hold "Quasi"
- press and release "Alt"
- release "Quasi"
- press "Quasi" (e.g. other hand)
- press and release "Ctrl"
- press "D"
- release "Quasi"
- press "D"
=> "Ctrl+D", "D" as a result.
```
The main advantage is reducing number of modifiers (no duplication for each hand) while keeping an ergonomic (hold modifiers on the opposite keyboard side).
@algernon that wold be fantastic to have such workflow in the Crysalis BTW.