keyboardio/Kaleidoscope #128: John Bumgarner's style key auto-repeat

Pitometsu, 2017-03-31:
The point: turn auto-repeat for symbol only after 3 press. E.g. for "=" key,
user should press "=", release "=", press "=", release "=", hold "=".

Here we have 2 advantages:
- we can't accidentally start key repeating just eventually holding key.
- we have two press intervals, so it's enough to approximate desired key repeating expected speed (yep, we able interactive choose key repeating speed).

Surely, it require turning off software key auto repeating in OS for best UX.

algernon, 2017-03-31:
This can be done with [TapDance](https://github.com/keyboardio/Kaleidoscope-TapDance), I believe, though, only for selected keys. Doing this for all keys would require quite a bit of bookeeping, as far as I see. May be doable, though.

algernon, 2017-06-02:
I *think* I have an interesting way this could be accomplished, that does not require turning off auto-repeat on the OS side:

We'd have a plugin, that looks for hold events, and instead of allowing hold to pass through, prevents it, unless the key held has been pressed two times before. We'd do this by counting releases. The plugin would not prevent the holding of modifiers, though.

Pitometsu, 2017-06-02:
Well, your approach have interesting OS compatibility advantage, and miss only minor feature - expectable repeating speed. It should work even with minimal auto-repeat begin delay on OS (only more then single keypress threshold). Reasonable compromise by default. Though would it possible to let user enable expectable repeating speed on demand (e.g. at compile time at least)?

algernon, 2017-06-02:
This approach would leverage the OS-side repeat, so if you press `a` twice, then hold it, it would start repeating after the OS threshold. Not immediately, though, but the delay isn't much.

Detecting a hold in firmware is trivial, and needs no timer at all, because we have access to the previous state, anyway. Let me give you an example, to make it clearer:

- We press `a`.
- The firmware notices the key is pressed, and records the key position, and registers the key.
- The firmware sends the USB report, and clears the report.
- The firmware enters the next scan cycle, notices that `a` is still pressed, and ignores it. As such, `a` will not appear in the next report.
- The firmware sends the USB report, without `a`, so it is not repeating. For the host, it looks like `a` was released.
- We release the `a` key.
- The firmware notices that we released the last pressed key, and increments the tap counter, and unregisters the key (for side effects by various plugins).
- The firmware sends the USB report.

All of this so far happens within about 15ms, or less. We do not need to use a timer, because at the third tap, we simply skip the ignore-this-key part, and the OS-side repeat will start its timer immediately. There is no extra delay introduced, except for the two taps needed to trigger the auto-repeat.

When the OS auto-repeat is at minimum, this still works, because auto-repeat can only trigger when two scan cycles report the same key pressed, twice. And with this setup, that only ever happens on the third tap. If it would start repeating when it first sees `a`, then you wouldn't be able to write a single `a`, unless you can tap and release it within ~5ms or less. That's *very* short.

Pitometsu, 2017-06-02:
Hm, about per-scanes approach, it's pretty elegant, however would work only in case of hardware keyswitcher's bounce protection. Otherwise, there should be software bounce protection and we would force to use timeout between bounce protection delay and OS auto-repeation begin delay.

Pitometsu, 2017-06-02:
And it would be nice to have firmware auto-repeating ability for those who don't like OS auto-repeating (but it have just a minor priority)