keyboardio/Kaleidoscope #137: Re-unify the project repos

wez, 2017-06-11:
# Rationale

I've found it frustrating to hack on the code.  I may be overly opinionated as my day job is building source control systems at a large company that loves monorepos.  However, I do feel like it is more difficult to work on this project than the big repos at work!

1. Many of the things I've wanted to change need to be coordinated across multiple projects
2. Keeping track of something that is one logical change locally means maintaining N+1 branches; 1 for each of the touched repos, plus an additional one in the `Arduino-Boards` repo that tracks them.
3. Github has no support for dependent PRs.
4. Travis-CI has no support for subrepos or dependent PRs.
5. Tracking and rebasing in the current model is pretty high friction; trying to rebase some work I have for the event dispatcher PR is taking way too long considering the magnitude of the changes

# Proposal

That we re-unite the projects that are referenced from Arduino-Boards into a single repo.

This script: https://gist.github.com/wez/81fcfdd90e5ae9cbc6fd614beacdf1fe
produced this branch as an example of what things might look like: https://github.com/wez/Kaleidoscope/tree/unify

# Implications

1. Changes across multiple libraries/plugins can now be submitted as a single PR
2. A single PR is easier for a human to review; fewer places to visit to understand the impact of that change vs. consuming multiple PRs with dependency or CI test results 
3. Travis-CI can be configured for just the main repo and can more easily be made to build and test PRs and commits, easing the review burden
4. The getting started instructions will be much simpler; check out a single repo and run `make` (or some script in the top level dir).
5. The history of the repo is preserved by the script, but github doesn't use the `--follow` flag in its history view.  Blame is correctly attributed, but you'll need to run some manual `git log --follow` commands to review history, should you need to review history.  It is possible for the migration script to use `git filter-branch` to rewrite the paths in the history to workaround this, but it is labor intensive to deal with conflicting paths in the history with that approach.  (I tried this and decided it was too much work to do without getting buy-in from the other contributors!)

# Concerns

## What does this mean for Arduino IDE users that are consuming this project?

Today, it is technically possible for someone to download and use one of the libraries and use it independently from the rest of the Kaleidoscope project.

With a unified repo this is still possible, but slightly more work: they just have to move the libraries subdirectory of interest out of the repo and put it in their own libraries folder.

Building a firmware project is easiest to do from the command line.  There are too many libraries to manually copy into the local sketch library folder today, so I don't believe that anyone would want to maintain those by hand.  If that is a flow we want to support, we can write a little script to symlink things to make it look right.

## Will this encourage poor layering/encapsulation choices when hacking on the code?

One argument for having separate repos is that it forces the code to be more modular.
Moving to a unified repo makes it easier to make a change anywhere in the repo but doesn't inherently lead to poor choices; they're still possible in both models.   For the reasons I mentioned in the rationale section, I am more inclined to put code into the wrong module just so that I can reduce the number of PRs I'd have to deal with.  Moving to a monorepo means that I can more easily prepare a series of small well-encapsulated commits.

## Other questions?

Please add them below as comments!

wez, 2017-06-11:
I also have a commit that layers on top of this that uses python to discover the projects and then executes the build via arduino-builder.  This is a cut down version of something similar I have in my own personal project.

You can see it in this branch: https://github.com/wez/Kaleidoscope/tree/build-bits

Running `./make --help` should be pretty self explanatory.   Why do that?   For my own projects, I'd like to pull in some external deps, and be able to build for other hardware platforms.  Python feels (subjectively, to me) to be easier to manage adding more complexity to the build without it becoming unwieldy.

obra, 2017-06-11:
>    Reverse link: [4]unknown
> References
>    Visible links

Hi Wez,

Thanks for the detailed writeup of your proposal. As I mentioned before,
I'm not a fan of the monorepo structure, especially for a project where
I  want to be encouraging a culture of third-party plugins, many of
which, I expect, will not be maintained by us.

One point I mentioned  on IRC yesterday when you weren't connected was
that in order for a monorepo structure to work with the Arduino IDE and
boards system, it needs to be structured precisely like the
Arduino-Boards repo. That structure isn't exactly...nice, but it's what
Arduino needs. 

Also, you mention that it's currently easiest to build the firmware from
the commnadline and suggest that using the Arduino IDE requires copying
libraries into their sketch folder. That is emphatically  not the case.
The whole point of the Arduino-Boards directoruy structure is that the
user can simply drop it into their arduino 'hardware' directory, pick
"Keyboardio Model 01" or another supported board from the dropdown in
the IDE and be good to go.

I understand your frustration working with changes impacting many
modules across repositories. That is, indeed, no fun. In general, I
expect sweeping API changes to be few and far between. I'd be more than
happy to  do the grundt work of applying your changes for this one from
a whole-project diff.

I'm happy to continue to chat about the repository structure issue, but
expect to be fairly nose-down for the next couple of weeks getting our
PVT run shipped.

chughes87, 2017-07-29:
Does a monorepo structure prevent people from being able to develop plugins all that much? I suppose plugin implementers have easy examples of how to develop an independent plug-in if the project is developed as a collection of independent plugin repos already. As a person that wants to contribute to this project, I'd feel more comfortable working on it if it was a monorepo.

obra, 2017-07-29:
>    Reverse link: [3]unknown

If the goal of a monorepo is to have a single thing you can check out
and be able to build the project, that'd have to be something of the
form of the Arduino-Boards repository. One problem with that is that the
Arduino-Boards repo is, essentially, a generated product for installing
Kaleidoscope on a specific couple of keyboards.

Can you talk more about _why_ you'd want it in a monorepo?

chughes87, 2017-07-29:
I'm using Arduino boards like you're describing. My concerns are the same as Wez's. Managing multiple PRs for a change sounds prohibitively bothersome. The cost seems to outweigh the benefits afaict. I admittedly don't know the project that well yet however.

obra, 2017-07-29:
On Sat, Jul 29, 2017 at 01:16:46PM -0700, Charles Hughes wrote:
>    I'm using Arduino boards like you're describing. My concerns are the same
>    as Wez's. Managing multiple PRs for a change sounds prohibitively
>    bothersome. The cost seems to outweigh the benefits afaict. I admittedly
>    don't know the project that well yet however.

If a single change touches multiple repos, then we're doing a bad job
with encapsulation.

In Wez' case, he was building us a new way to switch HID backends on the
fly. That meant touching every call to Mouse., Keyboard.,
ConsumerControl., and friends.

That exposed a pretty serious encapsulation issue - we were just using
the globals from the HID library anywhere and everywhere without any
sort of API in Kaleidoscope. 

The work to fix this finished up in
https://github.com/keyboardio/Kaleidoscope/commit/7919922036828d23cedd4b2b5e6ac7fb6e0dac7a

While it's always possible to contrive a scenario where this is an
issue, it should be incredibly rare for anything other than a core API change
or a build-system screwup to touch lots of plugins.

If there are other places where we're bleeding across, I want to know.

chughes87, 2017-07-29:
Okay sounds good

jyp, 2017-08-07:
Encapsulation requires to have a well-defined and stable interface for the "plugins". 
If there is no such interface, any plugin will be written against a specific version of the reposet. 

As far as I can see their is no such interface atm. If that is true the repo-split has more cons than pros.

algernon, 2017-08-08:
There is a reasonably stable interface (the `KaleidoscopePlugin` base class, and the hook functions). A bit underdocumented, but it is there. As we get closer to mass production, the interface will become more and more stable.

obra, 2018-10-09:
Astute readers will note that we haven't closed this ticket as WONTFIX. 

We've been struggling to figure out the right balance between multirepo and monorepo.

Because it's important to us that third-party plugins aren't treated as second-class citizens, I've been really hesitant to move Kaleidoscope to a scheme where all of our code is in a single repo and third-party code just...isn't.

Another of my worries is that a monorepo will encourage anyone who's building a kaleidoscope distribution to just fork the whole project, rather than to fork the bundle repo and add or branch individual plugins. 

It's also the case that some things such as bootloaders we source from upstream really do feel a lot more like they should be submodules, rather than snapshots checked into our repo.

As we've been starting to add more cross-architecture support to Kaleidoscope, I've also discovered that we end up having to replicate the libraries/ (plugins) directory underneath multiple architecture directories. It's my suspicion that we're going to end up with a disjoint set of libraries for various architectures. 

For the past little while, I've been starting to think that the right next step is to move most of the core plugins into the Kaleidoscope repo. This isn't a full monorepo migration, but I -think- that it would make life easier for most of the folks who find our multirepo setup frustrating and confusing.

It would have the effect of tightly coupling core plugins and the core. That can be convenient, but can encourage bad habits. It would also draw a stronger distinction between plugins that we believe to be stable and supported and those that aren't, while still making it possible to build a Kaleidoscope distribution for a device with bundled plugins and without forking the core. 

In the current world order, this would bring us from 60 submodules to 10:

avr/bootloaders/caterina
avr/libraries/AtmegaScanner
avr/libraries/FingerprintUSBHost
avr/libraries/HID
avr/libraries/Kaleidoscope
avr/libraries/KeyboardioHID
avr/libraries/KeyboardioScanner
avr/libraries/Model01-Firmware
build-tools
toolchain

algernon, 2018-10-09:
I think we can drop `FingerprintUSBHost` (only used by `HostOS`, and we're going to remove auto-detection - and thus, `FingerPrintUSBHost` use -, so there'll be no users left in our plugins), and `AtmegaScanner` too. The latter was only ever useful for the Shortcut port, it isn't generic enough to be reusable, so there's little point in keeping it there at this time. We can always re-add it, if we ever come up with a generic solution.

algernon, 2018-10-14:
So, as I'm working through preparing the plugins for a monorepo-merge, here's a status update, which also serves as a reminder for me where I am. For each plugin, the steps are the following:

- Create an `f/monorepo` branch
- Move sources to `src/kaleidoscope/plugin/` (and adjust namespacing accordingly), with only the main header left in `src/`. If need be, add backward-compatiblity `typedef`s and similar.
- Move the gist of `README.md` (everything but the Travis CI badges) to `doc/plugin/NAME.md`.
- If there's an `UPGRADING.md`, merge that into `doc/plugin/NAME.md`. It's better to have all documentation related to the plugin in one place.
- Commit these.
- Remove all files that we don't need for the merge, leaving only the examples, docs, and sources. `COPYING`, `CONTRIBUTING.md`, `README.md`, `.travis.yml`, `Makefile`, `library.properties` and others are going away.
- Any scripts go into either `extras/PLUGIN/` (if not ready to be used as-is), or `bin/` (if ready for general use).
- Commit these
- Push the branch

I mostly finished most plugins yesterday, but I'm going through the list again today to see where I'm at. My status is documented below. Plugins not marked done are either not started on yet, or are in progress (in which case they're in italics).

- [x] Colormap
- [x] Cycle
- [x] CycleTimeReport
- [x] EEPROM-Keymap
- [x] EEPROM-Keymap-Programmer
- [x] EEPROM-Settings
- [x] Escape-OneShot
- [x] FingerPainter
- [ ] *FocusSerial* (`UPGRADING.md` needs to be folded into the plugin docs)
- [ ] *GhostInTheFirmware* (need to write upgrade notes)
- [ ] Hardware-Model01
- [ ] Hardware-EZ-ErgoDox
- [ ] Hardware-Technomancy-Atreus
- [ ] Hardware-Virtual
- [x] Heatmap
- [ ] HIDAdaptor-KeyboardioHID
- [ ] *HostOS* (`UPGRADING.md` missing)
- [x] HostPowerManagement
- [ ] *Leader* (need to write upgrade notes)
- [x] LED-ActiveModColor
- [ ] *LED-AlphaSquare* (need to write upgrading docs)
- [ ] LEDControl
- [ ] LEDEffect-BootGreeting
- [ ] LEDEffect-Breathe
- [ ] LEDEffect-Chase
- [ ] LEDEffect-Rainbow
- [ ] *LEDEffects* (need to decide what to do about `TRICOLOR.md`)
- [ ] LEDEffect-SolidColor
- [x] LED-Palette-Theme
- [x] LED-Stalker
- [ ]  Macros
- [ ] *MagicCombo* (`UPGRADING.md` needs to be folded into the plugin docs)
- [ ] Model01-TestMode
- [ ] MouseKeys
- [ ] NumPad
- [x] OneShot
- [ ] *Qukeys* (need to write upgrading docs)
- [x] Ranges
- [x] Redial (but with breaking change)
- [ ] *ShapeShifter* (need to write upgrading docs)
- [ ] *SpaceCadet* (need to write upgrading docs)
- [ ] *Steno* (need to write upgrading docs)
- [ ] *Syster* (need to write upgrading docs)
- [ ] *TapDance* (need to write upgrading docs)
- [x] TopsyTurvy
- [x] TypingBreaks
- [x] Unicode
- [x] USB-Quirks

In most cases, the "upgrading docs" are about plugin classes moving to a `kaleidoscope::plugin` namespace. In these cases, compatibility is preserved through a `typedef`. I need to add deprecation warnings too.

algernon, 2018-10-14:
Additionally, to make the resulting monorepo layout less chaotic, most of the `Kaleidoscope` sources will move to `src/kaleidoscope` (and get properly namespaced), with only `Kaleidoscope.h` in `src/`. Thus, when all things are merged, `src/` will only have headers, and most things will be under `src/kaleidoscope/plugin/`, with a few things under `src/kaleidoscope`, `src/kaleidoscope_internal`.

This step is going to be a big one too.

algernon, 2018-10-14:
Oh, and the effects of moving to a monorepo: in the past, one of my worries was that if things are tightly coupled, we may end up with symbols not being optimized out when they should be. I seem to recall having issues related to that when I tried moving things around at some point. But as far as I can tell now, that's not the case, everything seems to be working fine, and I saw no increase in firmware size (rather, I saw a minor decrease instead).

I did have to turn `dot_a_linkage` off in `library.properties`, because having many subdirs confuses the linker with it enabled.

algernon, 2018-10-14:
Hm. I'll need to update some URLs in the docs: example and dependency URLs need to be updated, so that they don't point to the plugin repos anymore.

I'll add these as separate commits, so I don't have to rework every branch from scratch again. Thankfully, this can be very easily automated.

algernon, 2018-10-14:
Uhh. Turns out I'll need to do some more massaging here and there. Arduino gets confused by all the funky includes.

algernon, 2018-10-14:
With most essential plugins merged, the factory firmware comes out at 18 bytes *less* `PROGMEM`, and the same amount of data used. So far so good!

algernon, 2018-10-14:
The above three are fixed, gonna figure out why my firmware grew so big.

algernon, 2018-10-14:
TLDR: Prior to merging the plugins, I disabled `dot_a_linkage` in `library.properties`, because some examples failed to compile with it, and it didn't cause an issue with the factory firmware. But without that, the compiler has a harder time figuring out what can be dropped. Reenabling it, and fixing the buggy examples solved the size issue (and my sketch is smaller by 2 bytes now!)

algernon, 2018-10-16:
Hm, the -Hardware repos will be tricky, because they have examples, but we're not set up for compiling for multiple boards from the same repo. Will have to figure something out about that.

algernon, 2018-10-16:
Quick fix: teach `kaleidoscope-builder` to look for conf files in the example dir too.

algernon, 2018-10-17:
With #389, we're down to:

- [ ] Hardware-EZ-ErgoDox
- [ ] Hardware-Technomancy-Atreus
- [ ] Hardware-Virtual
- [ ] **HIDAdaptor-KeyboardioHID** (may or may not be able to merge it)

The Hardware ones should be possible to merge, as soon as I teach `kaleidoscope-builder` to allow building for multiple boards in one go (see above for limitations and whatnot).

HIDAdaptor will have to remain outside for now, because if we merge it, it will always get compiled, and that'd prevent using another adaptor. @obra had a great idea about that, but... that's going to be a separate issue.

Once the hardware plugins are in (at least the ErgoDox and the Atreus one), I think we can consider this issue resolved.

algernon, 2018-10-19:
With #427 merged, only `Hardware-Virtual` and `HIDAdaptor-KeyboardioHID` remain outside. I believe they're ok to have outside of core for now, until we can figure out a reasonable way to merge them in like the rest.

As such, I'd consider this issue resolved.

Do you agree @obra?

obra, 2018-10-20:
I think that's fair.
ᐧ

closed 2018-10-20 by algernon