keyboardio/Kaleidoscope #140: Verifying sketch fails with example firmware

jamesnvc, 2017-07-22:
After following the instructions listed here to install the Arduino IDE and set up the Kaleidoscope firmware, I get two errors when building my sketch (The sketch is currently a copy of the example firmware at https://github.com/keyboardio/Model01-Firmware).

The first error complains about their being two HID.h libraries, one in the arduino install directory and one in the keyboardio library. I tried moving the arduino one out of the way, but the second error stayed.

The second error indicates that it can't find the definition for various mouse-related things when compiling the MouseKeys plugin.

Full error message below:
<details>
Archiving built core (caching) in: /tmp/arduino_cache_613584/core/core_keyboardio_avr_model01_3f7379d53668113d472969c15c3bd10d.a
Multiple libraries were found for "HID.h"
 Used: /home/james/software/arduino-1.8.3/hardware/arduino/avr/libraries/HID
 Not used: /home/james/Arduino/hardware/keyboardio/avr/libraries/Arduino-HID
libraries/Kaleidoscope-MouseKeys/Kaleidoscope-MouseKeys.a(Kaleidoscope-MouseKeys.cpp.o): In function `MouseKeys_::scrollWheel(unsigned char)':
/home/james/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope-MouseKeys/src/Kaleidoscope-MouseKeys.cpp:110: undefined reference to `Mouse'
/home/james/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope-MouseKeys/src/Kaleidoscope-MouseKeys.cpp:110: undefined reference to `Mouse'
/home/james/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope-MouseKeys/src/Kaleidoscope-MouseKeys.cpp:110: undefined reference to `Mouse_::move(signed char, signed char, signed char)'
libraries/Kaleidoscope-MouseKeys/Kaleidoscope-MouseKeys.a(MouseWrapper.cpp.o): In function `MouseWrapper_::MouseWrapper_()':
/home/james/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope-MouseKeys/src/MouseWrapper.cpp:44: undefined reference to `Mouse'
/home/james/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope-MouseKeys/src/MouseWrapper.cpp:44: undefined reference to `Mouse'
/home/james/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope-MouseKeys/src/MouseWrapper.cpp:44: undefined reference to `Mouse_::begin()'
/home/james/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope-MouseKeys/src/MouseWrapper.cpp:44: undefined reference to `AbsoluteMouse'
/home/james/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope-MouseKeys/src/MouseWrapper.cpp:44: undefined reference to `AbsoluteMouse'
/home/james/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope-MouseKeys/src/MouseWrapper.cpp:44: undefined reference to `AbsoluteMouse_::begin()'
libraries/Kaleidoscope-MouseKeys/Kaleidoscope-MouseKeys.a(MouseWrapper.cpp.o): In function `MouseWrapper_::press_button(unsigned char)':
/home/james/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope-MouseKeys/src/MouseWrapper.cpp:44: undefined reference to `Mouse'
/home/james/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope-MouseKeys/src/MouseWrapper.cpp:44: undefined reference to `Mouse'
/home/james/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope-MouseKeys/src/MouseWrapper.cpp:44: undefined reference to `Mouse_::press(unsigned char)'
libraries/Kaleidoscope-MouseKeys/Kaleidoscope-MouseKeys.a(MouseWrapper.cpp.o): In function `MouseWrapper_::release_button(unsigned char)':
/home/james/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope-MouseKeys/src/MouseWrapper.cpp:44: undefined reference to `Mouse'
/home/james/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope-MouseKeys/src/MouseWrapper.cpp:44: undefined reference to `Mouse'
/home/james/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope-MouseKeys/src/MouseWrapper.cpp:44: undefined reference to `Mouse_::release(unsigned char)'
libraries/Kaleidoscope-MouseKeys/Kaleidoscope-MouseKeys.a(MouseWrapper.cpp.o): In function `MouseWrapper_::warp_jump(unsigned int, unsigned int, unsigned int, unsigned int)':
/home/james/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope-MouseKeys/src/MouseWrapper.cpp:44: undefined reference to `AbsoluteMouse'
/home/james/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope-MouseKeys/src/MouseWrapper.cpp:44: undefined reference to `AbsoluteMouse'
/home/james/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope-MouseKeys/src/MouseWrapper.cpp:44: undefined reference to `AbsoluteMouse_::moveTo(unsigned int, unsigned int, signed char)'
libraries/Kaleidoscope-MouseKeys/Kaleidoscope-MouseKeys.a(MouseWrapper.cpp.o): In function `MouseWrapper_::move(signed char, signed char)':
/home/james/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope-MouseKeys/src/MouseWrapper.cpp:110: undefined reference to `Mouse'
/home/james/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope-MouseKeys/src/MouseWrapper.cpp:110: undefined reference to `Mouse'
/home/james/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope-MouseKeys/src/MouseWrapper.cpp:110: undefined reference to `Mouse_::move(signed char, signed char, signed char)'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Keyboardio Model 01.
</details>

jamesnvc, 2017-07-23:
Solution seems to be to add `Mouse.begin(); and `AbsoluteMouse.begin();` to the setup in the sketch.

obra, 2017-07-23:
On Sun, Jul 23, 2017 at 02:42:28PM +0000, James Cash wrote:
>    Solution seems to be to add Mouse.begin(); andAbsoluteMouse.begin();` to
>    the setup in the sketch.

That's very, very strange. This was while trying to build the Model 01
firmware without modification, after selecting "Keyboardio Model 01"
from the "Board" menu in Arduino 1.8.3 on a mac?

jamesnvc, 2017-07-23:
I had made modifications to the keymap, but no other changes. The board was set to the Model 01 and this is using Arduino 1.8.3 on Ubuntu Linux.

jamesnvc, 2017-07-23:
Hm, but now I'm seeing if I clone the example repo and build it, it works.  I will try to nail down the exact differences.

obra, 2017-07-23:
On Sun, Jul 23, 2017 at 01:01:23PM -0700, James Cash wrote:
>    Hm, but now I'm seeing if I clone the example repo and build it, it works.
>    I will try to nail down the exact differences.

Conveniently, that's my dev environment. One thing that sometimes trips
me up on Ubuntu is accidentally launching the stock Arduino IDE, which
is still 1.0.x on Ubuntu.

jamesnvc, 2017-07-23:
Yes, I had previously had an issue with the old Arduino, back when this was called Akela 😛 

It looks like the subtle change I had made was `#include`-ing Kaleidoscope.h before MouseKeys - I can see that changing the order on the default firmware will cause the same behaviour I was seeing.

Perhaps there should be a comment to the effect that order of includes matters?

algernon, 2017-07-23:
I've seen similar issues too, and have `AbsoluteMouse.begin()` & `Mouse.begin()` in my sketch too. Saw linker issues when compiling via Makefiles (didn't check building from the Arduino IDE yet).

I don't really see why including `Kaleidoscope.h` before `Kaleidoscope-MouseKeys.h` would cause this issue. But that is a connection I have not made before, so it's at least something to start with! \o/

jamesnvc, 2017-07-24:
The more I try to figure out the cause, the less sense this makes, since 1) it seems repeatable that having Kaleidoscope.h before Kaleidoscope-MouseKeys.h will cause that error but 2) the first thing in MouseKeys is including Kaleidoscope.h.  Very puzzling!

algernon, 2017-07-24:
I think the issue is with linking order. If we include `Kaleidoscope.h` first, then the Kaleidoscope stuff will be linked first. If we include `Kaleidoscope-MouseKeys.h`, then those objects will come first. In the second case, the optimizer will not optimize out the `AbsoluteMouse.begin()` and `Mouse.begin()` calls from `Kaleidoscope-MouseKeys`, but will do so if it is not the first set of objects linked.

So the include order only matters because that is also the linking order, and the issue is with the linker removing the symbols, thinking they are unused.

I'll see if I can test this theory.

algernon, 2017-07-24:
Right. I just checked: linking order is the same as include order, and if I change the linking order to link `Kaleidoscope` last, the build appears to succeed (but I only did a quick hack by hand).

This is starting to sound like a linker / arduino-builder issue at heart.

algernon, 2017-07-24:
To further clarify: the fact that `Kaleidoscope-MouseKeys.h` includes `Kaleidoscope.h` does not matter as far as linking order is concerned. What matters is the order they are included in the Sketch, as far as I see, that is what `arduino-builder` will use to decide the linking order.

It's not that some symbols are not accessible to the compiler - they are, we are past the compilation step when the error occurs. It is the linker that throws up its hands, after trying to be too clever, and include order does not matter for the linker. Linking order does.

wez, 2017-07-24:
The symptoms sound like [SIOF](https://isocpp.org/wiki/faq/ctors#static-init-order)

jamesnvc, 2017-07-24:
@wes I don't think that's the issue in this case, as 1) MouseKeys doesn't contain static objects, 2) it is deterministic with the order of the includes, and 3) adding references to Mouse in the sketch.ino make the issue go away.

algernon, 2017-07-24:
I don't think this is SIOF, because even if I move the `Mouse.begin()`/`AbsoluteMouse.begin()` calls from the `MouseWrapper` constructor to `MouseKeys.begin()`, which is called explicitly, the linking stage still fails. And as far as I understand, SIOF happens at run-time, but we never get there in this case.

(Oh, and neither `MouseWrapper`, nor `Mouse`, or `AbsoluteMouse`, or `MouseKeys` are static - they are all externs. But my C++ isn't all that great, so there's a reasonable chance I'm not seeing the full picture here.)

algernon, 2017-07-28:
A bit of an update here: with recent changes to `Kaleidoscope` and `KeyboardioHID`, the original problem went away: the firmware now links. But alas, it does not work, the device nodes are not recognised under Linux.

If I put `#include <Kaleidoscope-MouseKeys.h>` first thing in my sketch, all becomes well. Still looking into what may be going wrong.

closed 2017-07-28 by obra