keyboardio/Kaleidoscope #143: Docs have some inconsistencies on flashing firmware

phildini, 2017-07-25:
I recently tried to flash the firmware and was super confused from the docs whether I should use the makefile or flash from the Arduino app. Some consistency and clarity would be super helpful, as would a note that holding down the `prog` button is necessary to successfully flash.

Thanks!

algernon, 2017-07-25:
You can use either the makefiles, or the Arduino app - both should work, and both are supported. Whichever you find most convenient. I suppose this should be highlighted somewhere too.

phildini, 2017-07-25:
@algernon yep! filed this issue at @obra's request. Once I knew the steps the process worked great, but there are a couple gotchas that make things tricky for the beginner right now. 🙂

closed 2017-08-27 by algernon