keyboardio/Kaleidoscope #145: Going in and out of numpad mode often sets the led color to red

chughes87, 2017-07-26:
Regardless of what the previous setting was, the keyboard will be in the red state after going in and out of numpad mode. This doesn't 100% consistently happen though. It seems to happen the first time after logging back into my computer but not again until I log out of my computer and come back to my desk later. So the problem may be with persisting the selected color scheme during computer sleep states?

obra, 2017-07-26:
On Wed, Jul 26, 2017 at 08:31:46PM +0000, Charles Hughes wrote:
> Regardless of what the previous setting was, the keyboard will be in the
>    red state after going in and out of numpad mode. This doesn't 100%
>    consistently happen though. It seems to happen the [...]

There are patches that landed a couple days ago to fix this. Are you
currently running stock firmware or have you reflashed in, say, the last
36 hours?

chughes87, 2017-07-26:
oh I haven't reflashed yet. I will try this later and let you know if it resolves the issue. Thanks!

obra, 2017-07-26:
On Wed, Jul 26, 2017 at 08:39:13PM +0000, Charles Hughes wrote:
>    oh I haven't reflashed yet. I will try this later and let you know if it
>    resolves the issue. Thanks!
> 

Great. Thanks.

chughes87, 2017-07-27:
This was resolved after flashing 👍

closed 2017-07-27 by chughes87