keyboardio/Kaleidoscope #149: ;/: button doesn't turn red in numpad mode

chughes87, 2017-07-27:
https://photos.google.com/share/AF1QipPf1viYX5Lu4AFHktXvTPmOho_xXQ0rXqsaLkPo9qWpbfYJoy2NZdbN5Eqg6Rw17g?key=aTNnelhTWk91ajJsUk9tZUtTQ0lpdWF1XzZDelZ3

chughes87, 2017-07-27:
I suppose the same applies to -/_/| as well

obra, 2017-07-27:
>    Reverse link: [3]unknown
> References
>    Visible links

The reason they don't light up is that they aren't actally numpad keys.
I'm 100% open to modifying the keymap to make them numpad keys, though
it's too late to change the layout cards.

chughes87, 2017-07-27:
They may not be numpad keys in the traditional layout but they are keys that are affected by the numpad mode (currently at least). I'd say if the key behaves differently in numpad mode then it should change colors to indicate that.

obra, 2017-07-27:
Oh, then that is, indeed a bug!
 
@algernon - any idea?

algernon, 2017-07-28:
As someone else found out on IRC as well, it is [this](https://github.com/keyboardio/Kaleidoscope-Numlock/blob/401e9326a050316b54777fe0fab7273aa983c44d/src/Kaleidoscope-Numlock.cpp#L25-L26) part of the code that decides what gets colored, and what does not.

I could change that code to color everything that changed between layers, save for layer switching keys. That would highlight all the things it needs.

I'll prep a patch, and we'll see how that works.

closed 2017-07-28 by obra

chughes87, 2017-07-28:
@algernon It looks like your fix lights up the `any` and `prog` keys. Was this intentional?

obra, 2017-07-28:
>    Reverse link: [4]unknown

>    [1]@algernon It looks like your fix lights up the any and prog keys. Was
>    this intentional?

Ha. It lights up "Any" because Any has a special meaning on that layer
(That's unrelated to the numpad) 

That's where we hid the "print out the keyboard version" macro.

I have no idea why prog is lit up.

I'd be amenable to moving out the behavior on "Any", just to make life
easier.

algernon, 2017-07-30:
Uhh, not sure why `Prog` lights up, either. I'll open a separate issue about that, and look into it.

As for `Any`: that was intentional, because even though it is not a numpad key, it changed behaviour. It would be easy to amend the code to ignore macros too, or only highlight "normal" keys, if that would result in less surprising behaviour.

algernon, 2017-07-30:
keyboardio/Kaleidoscope-Numlock#6 should address the `Prog` & `Any` highlighting issue.