keyboardio/Kaleidoscope #150: Releasing `fn` before key enters 2 characters

obra, 2017-07-28:
@bergey writes: 

On the PVT with default firmware, on Windows 10, releasing the fn key while still holding one of uiopprints both characters - {u, [o- and so forth. I find I roll my hand in a way that I do this frequently. Does anyone want this behavior

obra, 2017-07-28:
I think I agree that it's not desirable, but it does seem to "work like Shift does"

I feel like we can probably do better on this.

bergey, 2017-07-29:
It seems to me that when I release Shift while still holding a letter, the lower-case letter begins to repeat after a pause.  I'd guess the same pause time as when I press & hold a letter without involving Shift.  This is the same with the Model 01 & with other keyboards; and on Linux as well as Windows.

With the Fn pairs, I can't detect a pause, and haven't managed to release the letter fast enough after the Fn key that I don't get a second character.  I think I'd be happy with either behavior:
- After releasing Fn, don't send keycodes for currently held keys until they are released
- After releasing Fn, pause briefly before sending keycodes

Thanks for moving this issue to the appropriate repo.

algernon, 2017-07-30:
I too agree, that the current behaviour is undesired, and a similar issues is plaguing TapDance (keyboardio/Kaleidoscope-TapDance#9) - or at least I feel it is similar. The easiest would be to mask out keys when releasing `Fn`, and make that a generic thing, so it can be used by other plugins.

Something like `maskHeldKeys()`, which would temporarily ignore all key events for keys that are held, and remove them from the mask when they are released. This can be done with a pair of `uint32_t` bitmasks. OneShot already implements [something similar](https://github.com/keyboardio/Kaleidoscope-OneShot/blob/42faebc87339165c4a2240240771a6d044e4baa4/src/Kaleidoscope/OneShot.cpp#L128-L132), I can pull that out into core.

I'll see if I can reproduce the problem, and if the suggested masking helps - if it does, I'll prep a PR.

obra, 2017-08-04:
@bergey I've merged @algernon's two proposed patches to deal with this issue. I'd love to know if a fresh update and build resolves this issue.

Thanks!
Jesse

ToyKeeper, 2017-08-05:
Just leaving a +1.  I was going to report this issue, but it seems I'm a week late.  :)
I tend to end up with extra unwanted characters after releasing Fn, so it's nice to see that a fix is already in progress.

algernon, 2017-08-05:
I just tested this again, while working on keyboardio/Kaleidoscope-OneShot#10, and the masking did not solve the issue by the looks of it. `Fn` held, then `u` held produces repeating `{`, but as soon as I release `Fn`, I get `{u` printed, then `u` repeated.

algernon, 2017-08-06:
Right, the problem is that the masking wasn't applied properly, and that part is fixed in on [Kaleidoscope-Hardware-Model01 master](https://github.com/keyboardio/Kaleidoscope-Hardware-Model01/commit/0a566eb58fb626e2413a31aceefaf6e8cb51a169).

ToyKeeper, 2017-08-06:
I don't suppose there's any chance of not changing the layer on Fn release?  So, if Fn-u produces repeating {'s, and the user releases Fn but continues to hold u, it would continue to repeat { instead of u?

That's how my HHK works, at least.  It seems like a good way to do it.

Other mods switch on release, but not Fn.  So, "+Ctrl, +e, -Ctrl" produces repeating 'e'.  Same with Shift, or Alt, or Hyper, or any mod handled by the computer's OS.  But Fn is special since it's handled by the keyboard's firmware.  If the Fn layer has an action defined for a key (instead of falling through), the underlying layer isn't activated on Fn release.

Basically, the active layer doesn't change until a key is released.  The event sent to the host for a given key is set on keypress, and doesn't change until the key is released.  This would also mean that, if the user presses left Fn for layer 1, presses 'u', then presses right Fn for layer 2, the key would continue to make '{' events even if layer 2 has a different mapping for it.

algernon, 2017-08-06:
Repeating `{` in the scenario you describe would not be trivial to implement, at least at a guess. I suppose we could provide a knob or a plugin to have that behaviour. Can you open an issue about it? :)

obra, 2017-08-07:
On Sun, Aug 06, 2017 at 08:29:58PM +0000, Gergely Nagy wrote:
>    Repeating { in the scenario you describe would not be trivial to
>    implement, at least at a guess. I suppose we could provide a knob or a
>    plugin to have that behaviour. Can you open an issue about it? :)

Alternatively, we could define the momentary layer system as only
toggling off when NO keys are pressed. That seems like it'd get us
behavior that's much closer to user expectations.

algernon, 2017-08-07:
I'll see what I can do.

ToyKeeper, 2017-08-07:
I'm running the code with Layer.repeat_first_press now, and it seems to be working great so far.  I set it to true in my sketch and I haven't encountered any issues with it.  It's nice being able to keep repeating something from another layer without holding the layer key, like arrow keys and mouse wheel actions.

gedankenexperimenter, 2017-08-09:
It looks like this issue should be closed, just like #158

algernon, 2017-08-09:
Closing this - if the issue still persists, or resurfaces, please reopen!

Thanks a lot for reporting it @bergey, it kickstarted a very useful discussion, which ended up with not only fixing the issue, but with a much faster scan cycle, and less surprising momentary layer behaviour too. \o/

closed 2017-08-09 by algernon

bergey, 2017-08-12:
Sorry for the late response.  I confirm that this bug *was* fixed, and the behavior of 491dca3 works well for me.  Thanks for the discussion & code changes.

I am very amused that it's broken again, per #173 =)