keyboardio/Kaleidoscope #153: Mute key not working

chughes87, 2017-07-31:
Volume up and down work fine. The mute button doesn't seem to do anything in OSX 10.12.5

algernon, 2017-08-06:
It appears that mute doesn't work under Linux, either.

obra, 2017-08-07:
>    At least on Linux, using Key_Mute instead of Consumer_Mute fixes the
>    issue. Can you check if it works on OSX too?
>
>    â

FWIW, I've pushed this change to Model01-Firmware for testing.

ailogic, 2017-10-27:
win 10, confirmed, Key_Mute does not work, Consumer_Mute does not work.

cdisselkoen, 2017-11-04:
https://community.keyboard.io/t/windows-mute-volume-key/581 has some more details and diagnosis of the problem

algernon, 2017-11-24:
I *think* this should be fixed once #242 gets merged.

obra, 2017-11-24:
I'm pretty sure that the current state of master (and the MP2 firmware)
already do have this fixed.

algernon, 2017-11-24:
...and you are right. Lets close this one then!

closed 2017-11-24 by algernon