keyboardio/Kaleidoscope #164: We need a tutorial explaining basic keymap customization

obra, 2017-08-09:
This tutorial document, aimed at the end-user of a keyboard Kaleidoscope firmware, should step a user through how to modify their firmware's default .ino sketch to change the key layout. It does not need to address advanced features such as sticky modifiers, one shot keys or LED effects.

It should reference, but not include Arduino and hardware-support library installation documentation.

This issue partially replaces #162

obra, 2017-08-09:
From https://community.keyboard.io/t/preparing-plugins-for-1-0-0-looking-for-documentation-reviewers/263/19

> Should one make a copy of Model01-Firmware? Make a new repository from scratch?
> What files should be modified to make changes? And how?
> The plugin docs (such as they are) refer to a thing called a “Sketch” – what is that?

Jennigma, 2017-10-13:
I think this is done, at least in first draft:

https://github.com/keyboardio/Kaleidoscope/wiki/Edit-A-Keymap

obra, 2017-10-13:
Yay! Thank you. please feel free to resolve doc tickets when the docs are
drafted and posted :)

closed 2017-10-13 by Jennigma