keyboardio/Kaleidoscope #165: We need an overview document describing available, "supported" Kaleidoscope modules.

obra, 2017-08-09:
This document should include end-user friendly capsule descriptions of known, moderately-trusted Kaleidoscope plugins, as well as links to those plugins usage documentation.

It should include author and maintainer information for each plugin.

We can probably steal the content from https://asylum.madhouse-project.org/blog/2017/03/10/rose-tinted-glasses/ to get started. (With @algernon's permission)

This issue partially replaces #162

algernon, 2017-08-09:
You can steal that content, it's under CC-BY-SA 3.0. I'm willing to relicense it to something else reasonable if another license would fit the result better. (GPL, GFDL come to mind as example licenses I'd be happy with.)

obra, 2017-08-09:
>    Reverse link: [3]unknown
> References
>    Visible links

CC-BY-SA seems totally reasonable for docs.

algernon, 2017-08-16:
What if we made this automated? Go through `Arduino-Boards/libraries`, look at the `README.md` and `library.properties`, and if the README marks the plugin "stable", include it in the collection. If all our READMEs would follow a similar structure (they mostly do, with few exceptions), that would allow lifting out the introductions programmatically.

The upside of this is that it would be always up to date (and we can add travis badges, last commit info, version number, etc). The downside is that we need to mark plugins stable / experimental / obsolete / whatever. Another benefit of doing it automatically is that we could have a "Supported plugins", an "Experimental Plugins" and perhaps an "Obsolete / Broken plugins" section too, all generated from the same source by the same tool.

We could also add "Community plugins" by going through the repos on GitHub [tagged with `kaleidoscope-plugin`](https://github.com/search?q=topic%3Akaleidoscope-plugin+org%3Akeyboardio&type=Repositories), and collect info from those too.

algernon, 2017-08-16:
It could be a bit like [`travis-status`](https://github.com/keyboardio/Kaleidoscope-Plugin/blob/master/tools/travis-status) in `Kaleidoscope-Plugin`.

algernon, 2017-08-16:
If we end up with an automatic method, and we write it in JS (as a library) or ClojureScript, that would allow me to reuse the same thing in Chrysalis too. ;)

obra, 2020-06-08:
https://kaleidoscope.readthedocs.io/en/latest/#plugins solves this, I believe.

closed 2020-06-08 by obra