keyboardio/Kaleidoscope #166: Raid Algernon's blog posts for tech notes

obra, 2017-08-09:
https://asylum.madhouse-project.org/blog/ contains numerous technical notes about Kaleidoscope's design. When appropriate, with permission and credit, we should import that content into Kaleidoscope's documentation.

Jennigma, 2017-10-15:
I'm happy to pick this up.  Suggestions on how best to manage attribution and licenses welcome.

algernon, 2017-10-15:
For most of the things on my blog, I think lifting them as-is is not all that useful. Reworking them would yield better results. This makes it much harder to credit & license things, so... how about I give y'all permission to use any keyboard-related post form my blog as if it was in the public domain? No need to care about any license then.

As for credit, if you stick a link back to my blog somewhere, that's great. [This](https://asylum.madhouse-project.org/blog/tags/keyboardio/) would be a reasonable link, I suppose. Perhaps having an "Additional resoures" page or similar would make sense, and the previous link would have a nice place there. That way we don't have to litter pages that use content from my blog with annoying referrals, license blurbs and credit. :P

Jennigma, 2017-10-16:
I can add a footer for all the pages that goes to a credits page, and an author attrib at the bottom of pages wouldn't be too much clutter. I need a place to put credits for the icons on the New Home page, assuming they pass muster, anyway. I thought github would show the alt text on hover and that would work for attrib, but it doesn't.

Yes, I agree that your blog entries will require rework, but I'll learn stuff. Worthy trade. :-)

algernon, 2017-10-16:
A footer with links to a credits page is a good idea!