keyboardio/Kaleidoscope #167: Document Kaleidoscope hardware backend API

obra, 2017-08-09:
We need to formalize the API exported by Kaleidoscope hardware implementations. Currently, the gold standard is Kaleidoscope-Hardware-Model01. @algernon is working on a new base class for that to inherit from. Once we have that, every API method should have doc on use case, calling convention and valid inputs/outputs.

algernon, 2017-08-09:
Since I'm working on the base class anyway, I'll assign this to myself, so I remember to write docs too.

algernon, 2017-08-16:
I have started working on [Kaleidoscope-Hardware](https://github.com/keyboardio/Kaleidoscope-Hardware), a base class for all hardware plugins. It has at least some docs, but will likely need some improvements.

cdisselkoen, 2017-11-04:
This can probably be considered resolved by https://github.com/keyboardio/Arduino-Boards/commit/acaa674f75a525f0b553d32ecf7ceab8892ff669

cweagans, 2017-11-21:
Kaleidoscope-Hardware is a good first step, but I'd also like to see some information about the high-level stuff that one needs to do in order to port this firmware to other hardware. I'm interested in porting Kaleidoscope to the Ergodox Infinity, but I have no idea where to start.

PR #504 (2018-12-18): Implement a kaleidoscope::Hardware base class
This is basically the old Kaleidoscope-Hardware plugin pulled in, and made into a base class. The purpose is documentation (#167) mostly. The base class has default implementations for all methods (most do nothing), and can be overridden by derived classes. Since the hardware object is always going to be called through `KeyboardHardware`, which has the exact type of the hardware, we do not need to use virtual functions. The derived class will just shadow the default implementations.

It also moves `detachFromHost`/`attachToHost` to the base class,  [... cut]
closed 2018-12-18 by obra