keyboardio/Kaleidoscope #176: Consumer control keys don't repeat

obra, 2017-08-13:
I don't remember _why_ doing this is tricky, but I recall that it is tricky. And what works fine for linux is totally broken on OSX

algernon, 2017-08-13:
It is tricky, because you can't rely on the OS doing the repeat, because they handle held consumer keys differently. Linux has similar behaviour as for normal keys, OSX goes bonkers. So we'd need firmware-side repeat.

See 8bb9bb37ca76caa1f79ca615b1b921576e51e2ad for example, which changed the repeat behaviour introduced in #121.

eltang, 2017-08-13:
I'm not too familiar with the internal workings of Kaleidoscope yet, but I know other non-Arduino firmware is able to send consumer scan codes the exact same way they send keyboard scan codes without any problem.

obra, 2017-08-13:
I suspect the issue is actually similar to the repeated key report issue we fixed on the keyboard/keypad side.

The correct fix is probably in KeyboardioHID. 
Mostly my cautionary note was that a fix needs to be tested on multiple platforms.

eltang, 2017-08-13:
I took a quick look at the code in KeyboardioHID and I think the problem is that multiple calls to ConsumerControl.press with the same scancode will quickly fill up the report with multiple copies of that scan code.

algernon, 2017-11-24:
I *think* I have a way to fix this for all OSes: teach `ConsumerControl` to not send a report on every press or release, and then teach Kaleidoscope to `press` the consumer control key whenever it is pressed, and clear the report every cycle, just like we do for the keyboard.

This appears to work on Linux. Sadly, that's the only platform I can test at the moment. Will open PRs for testing purposes.

PR #251 (2017-11-29): Fix ConsumerControl repeating
Instead of sending a press & release for the consumer key when the physical key is released, send a press each cycle it is held, a report along with the keyboard report, and clear the consumer report each cycle too.

This will prevent these keys getting stuck, or sending multiple presses in the same report.

Fixes #176.

Depends on keyboardio/KeyboardioHID#13.
commit be49df28 (2017-11-29): Fix ConsumerControl repeating

Instead of sending a press & release for the consumer key when the physical key
is released, send a press each cycle it is held, a report along with the
keyboard report, and clear the consumer report each cycle too.

This will prevent these keys getting stuck, or sending multiple presses in the
same report.

Fixes #176.
closed 2017-11-29 by obra