keyboardio/Kaleidoscope #178: API: Timer helpers

algernon, 2017-08-16:
We have a lot of plugins that use timers for one thing or the other, and they all implement time diffing themselves, with no attention paid to timers wrapping around.

We should have helpers in the core firmware that make it easier to deal with timers: read a timer, check the time elapsed, etc. To make the API useful, go through all the timer-using plugins and see how they use timers.

gedankenexperimenter, 2017-12-11:
I used a timer in Qukeys, and decided that a premature timeout window of less than one second every 50 days (if the keyboard stays on that long) was an acceptable risk, and that it wasn't worth the extra instructions on every cycle to guard against it. Are there any plugins that use timers that are much longer than ~one second?

algernon, 2017-12-11:
[OneShot](https://github.com/keyboardio/Kaleidoscope-OneShot) uses a 2.5 second timeout by default, and that's the longest timer I could find quickly.

algernon, 2017-12-11:
Mind you, the guard is useful not to prevent a premature timeout, or to prevent failing to time out, but to prevent crashing behaviour when the timer overflows (which might have been the case in keyboardio/Kaleidoscope-LED-Stalker#8).

gedankenexperimenter, 2017-12-11:
Can you explain how a timer overflow could cause that? I thought it would just wrap and start over from zero, but I don't see how that would cause an out-of-bounds memory access (if that's what you think was going on there). Does it do more with the return values of `millis()` than just comparison operations?

algernon, 2017-12-11:
On second thought, no, it does not.

gedankenexperimenter, 2017-12-17:
Two things that occurred to me recently:

* TypingBreaks uses a very long timer (but it honestly doesn't seem like a big deal if it has an overflow problem; you could just power cycle the keyboard — which is probably something a naïve user would try)
* The overflow problem is solvable with some extra computation. Instead of testing:
```
millis() > end_time
```
…test:
```
millis() > (start_time + timeout)
```

So, I do think a timer helper would be good to have. I'll probably write one at some point, if nobody else does it first.

gedankenexperimenter, 2019-05-06:
I think I will actually do it for real this time! (This comment is intended as a reminder for me to write the code for this next time I'm in front of a computer.)

gedankenexperimenter, 2019-05-06:
I've been looking around, and found several timer instances where plugins are simply comparing against an end time, and therefore will have premature timeouts on integer overflow. There may be other places where a few bytes can be saved by using a smaller integer type (2 bytes instead of 4), and there are still lots of plugins calling `millis()` directly.

PR #641 (2019-06-10): Add timer helper function
This adds a function to help plugins that use timeouts to efficiently and correctly check those timeouts by comparing the elapsed time to a timeout interval without errors due to overflow. It uses the same time value as the existing `millisAtCycleStart()`, which should be used to set the "start" timestamp.

Closes #178.
commit 71fd49e6 (2019-06-10): Add timer helper function

This adds a function to help plugins that use timeouts to efficiently and
correctly check those timeouts by comparing the elapsed time to a timeout
interval without errors due to overflow. It uses the same time value as the
existing `millisAtCycleStart()`, which should be used to set the "start"
timestamp.

Closes #178.
closed 2019-06-10 by obra