keyboardio/Kaleidoscope #180: 3 key sometimes double taps on single keystroke

chughes87, 2017-08-17:
Occasionally when I press the `3` key I get two 3's on the screen when I only pressed the button once. This is SUPER bothersome when trying to enter passwords with numbers in them. Below I will press the three key and enter a few times:
3
3
3
3
3
3
3
3
3
3
3
33
33
3
3
3
3
333
33
3
33
33
3
3
3
33
33
3
33
3
3

chughes87, 2017-08-17:
This doesn't seem to happen with any other key

obra, 2017-08-17:
Hi Charles,

This is a hardware issue, though there might be a few possible causes. Depending on your hardware comfort level, there are a few possible ways to resolve it.

**1. The most likely cause is a poorly soldered keyswitch. It's a known issue with some of the PVT keyboards. It's relatively easy to diagnose, though it requires a screwdriver.**

Disconnect the left half of the keyboard from your computer and from the right half.
Flip it over.
Remove the screws on the bottom of the keyboard.
Flip it back over.
pull the wooden enclosure straight up off the plastic baseplate.
Unscrew the PCB from the baseplate.
Flip the PCB over.
You're looking for, I believe, key R0C3. If the solder joints don't look perfect or you seem to be missing a some solder, please snap a photo for me and mail it to me at jesse@keyboard.io. From there, we can discuss options.

**2. It could be a faulty keyswitch.**

We've seen one case of a switch that was misassembled. In that case, the key was triggering twice on keypress. Once at the top of the cycle and one at the bottom. 

One way you _might_ be able to detect it is to use our factory test mode. 

1. Tap Prog + LED + Left FN. 
2. The keyboard will turn red.
3. Tap Left FN
4. The keyboard will turn green.
5. Tap Left FN
6. the keyboard will turn blue.
7. Tap Left FN
8. The keyboard will turn white
9. Tap Left FN
10. The keyboard will flash through a rainbow, ending up red.
11. If you press a key, it will turn green while actuated and blue when released.

If you have a faulty key, you're likely to see it turn green, then blue, then green again as you press it down slowly. The pattern is likely to be the same on repeated presses.

If you have a short caused by a badly soldered diode or switch, you're likely to see the keyboard flash between green and blue when you press the key all the way down and either push the key back and forth or whack the circuit board with your hand.

**3. It could be a badly soldered diode.**

The PCB assembly factory our factory used for the PVT keyboards screwed up the diode soldering pretty badly and they ended up needing to be reworked by hand. There's a chance that one of them wasn't well soldered and is shorting out.

If the test procedure above doesn't work, It's pretty much impossible for you to diagnose this locally (or to differentiate it from # 2. If this is the issue, we pretty much have to set up an RMA, either for your whole keyboard or for the left PCB + keycap assembly. If that's the case, drop me a note at jesse@keyboard.io and we'll get it set up.

There is a potential software workaround that should reduce key chatter. But it's a workaround, not a proper fix. I'm not 100% sure it'd help and I can't recommend it. You'd want to apply this change to Kaleidoscope-Hardware-Model01 and rebuild your firmware.

jesse@szx:~/git/Arduino-Boards/libraries/Kaleidoscope-Hardware-Model01 ((71a288c...))$ git diff
diff --git a/src/Kaleidoscope-Hardware-Model01.cpp b/src/Kaleidoscope-Hardware-Model01.cpp
index d23e63b..7f33f42 100644
--- a/src/Kaleidoscope-Hardware-Model01.cpp
+++ b/src/Kaleidoscope-Hardware-Model01.cpp
@@ -60,6 +60,8 @@ void Model01::setup(void) {
   rightHandState.all = 0;
 
   TWBR = 12; // This is 400mhz, which is the fastest we can drive the ATTiny
+  leftHand.setKeyscanInterval(100);
+  rightHand.setKeyscanInterval(100);
 }

obra, 2017-08-17:
Also, since you're local, I could pop into the city with a replacement later today if that'd be easier for you.

chughes87, 2017-08-17:
Looks like it's a solder joint issue. Not software

closed 2017-08-17 by chughes87