keyboardio/Kaleidoscope #184: feature request: disallow flashing until password is typed on keyboard

ian-kelling, 2017-09-18:
This is a security feature. Make it so you can setup a password, then 
when you type the password, it allows you to flash for the next minute
or something. Otherwise, flashing will fail. 

One especially troubling security issue is persistent malware that alters the flashing
process so that it embeds itself into any new software flashed to the device.

obra, 2017-09-18:
Hi Ian,

The current implementation only allows flashing over USB _while the prog
button is being held_. Additionally, because of the way the AVR chip inside
works, a flashed firmware can not overwrite the bootloader, nor can it
influence the bootloader's operation.

As far as I know, the attack you describe with malware "infecting" new
firmware is impossible unless the bootloader is modified. To do that, you'd
need to unscrew the keyboard and reflash the bootloader using a hardware
programming device.

Best,
Jesse

ian-kelling, 2017-09-18:
Thx. My lack of understanding.

closed 2017-09-18 by ian-kelling