keyboardio/Kaleidoscope #194: Deprecating things & lifetime of deprecated stuff

algernon, 2017-10-05:
As has been the case in the past, so it will inevitably be in the future: sometimes we will find a better way to do something, and want to deprecate an old API. We do not have a policy for this yet, but we should.

The deprecation part is the easier thing, this issue is more about how long we keep deprecated APIs?

algernon, 2017-10-05:
My gut feeling is that unless there is a good reason to deprecate and remove things faster, about half a year is plenty. Half a year may also sound plenty, but it gives enough time for our users to update their sketches & plugins at their leisure.

When we deprecate things in a way that is still compatible, the burden of carrying that extra code around is fairly small, so I see no reason why we should remove them fast. Heck, we used to have deprecated interfaces that weren't ever used by any plugin, and we only removed them yesterday.

obra, 2017-10-05:
The counterpoint is how we broke numpad backcompat with 0 notice ;)

algernon, 2017-10-05:
We broke quite a few things with 0 backwards compat in the past... hopefully we can avoid doing so in the future =)

algernon, 2018-10-19:
Recent practice is that we keep deprecated interfaces for at least three months. I think that's reasonable, and since it's already existing practice, I think this issue can be resolved.

closed 2018-10-19 by algernon