keyboardio/Kaleidoscope #207: Document the list of plugins

Jennigma, 2017-10-10:
craigdissel suggested in the forums creating a master list of plugins. I intend to do this in the docs somewhere, and am creating myself an issue so I don't lose track of the idea.

reference:

https://community.keyboard.io/t/dynamic-macro-plugin/402/8?u=jennigma

quote:

Another thing I’d totally like to see is a master list of available plugins (ideally, all plugins in all three of these categories). This goes regardless of which plugin model we use. Even a list with plugin name, plugin author/maintainer, single-line description of the plugin, and link to install would be great; most plugins already have all this info in their library.properties file, so we’d just need to scrape that. The resulting list should be distributed with Kaleidoscope core, or perhaps (to allow more frequent updating) we could just distribute a script or something to pull the latest list. In any case it should be very accessible to end users with little knowledge of Arduino or command-line tools.

algernon, 2017-10-10:
What would be fantastic is if we didn't have to maintain the list by hand, but had a tool that generated one for us. If that tool would be a piece of JavaScript, we could stick it onto GitHub pages, and mission accomplished.

- We can check `Arduino-Boards` via the GitHub API, and see what plugins it pulls in.
- We can search GitHub for `topic:kaleidoscope-plugin` to find any plugins tagged with that, but perhaps not part of `Arduino-Boards`.
- We can extract & parse the `library.properties` file, and even the READMEs.

While a manual list would be easier to consume perhaps, it would also be a lot harder to maintain. Unless, of course, we make it a WiKi page! That may be the best, most scalable solution.

Jennigma, 2017-10-10:
i think the docs should probably get broken into a wiki in the medium term, just starting with a monolithic doc so I can figure out where the natural breaks are as it develops.

cdisselkoen, 2017-10-10:
Oh, turns out this is largely a duplicate of #165.

Jennigma, 2017-10-10:
great minds!

Jennigma, 2017-11-05:
I took a VERY quick pass at this, giving names and descriptions for the core plugins and LED effects. It ain't pretty, but something is better than nothing. :-)

obra, 2020-06-08:
https://kaleidoscope.readthedocs.io/en/latest/#plugins should solve this

closed 2020-06-08 by obra