keyboardio/Kaleidoscope #211: where to store images for docs?

Jennigma, 2017-10-14:
I would like to be able to add images to the docs. I see that @obra used his own repo for the images in the bootloader page of the wiki:

https://github.com/obra/KeyboardioFirmware/wiki/images/

I prefer not to believe that my repo will persist indefinitely, and if there is a way for me to add images to the main wiki/images/ directory, I haven't found it. I would welcome help figuring it out.

I could create a docs/images directory and reference that.

Suggestions?

The proximal problem is sharing images of the Dvorak and Colemak layouts. I have saved nice .png's from the .svg, just need a place to put them.

obra, 2017-10-14:
You should be able to check out the Kaleidoscope wiki as a git repo and add
the images as files to it.

Jennigma, 2017-10-14:
got it working.

closed 2017-10-14 by Jennigma