keyboardio/Kaleidoscope #214: Document adding modifiers to a key right on the keymap

obra, 2017-10-16:
https://community.keyboard.io/t/programming-any-key-to-send-a-chord/426/5

Jennigma, 2017-10-17:
Got it.

grufelous, 2019-05-12:
Hi! I'm interested in resolving this issue under Season of Docs. I'd love a few pointers on how to get started with this issue, in particular about any specific requirements.

algernon, 2019-05-28:
Hi @grufelous! Apologies for the terribly late reply for which I have no good excuse.

A starting point would be the forum link in the opening comment ([this one](https://community.keyboard.io/t/mapping-a-single-keypress-to-a-chord/426/5)). You can also look at the [Planck example sketch](https://github.com/keyboardio/Kaleidoscope/blob/ac434360c755f6d19a7234f903f62b24771d5b5a/examples/Devices/Planck/Planck.ino#L103) for an example about how to add modifiers to a key in the keymap.

It's a reasonably simple thing, but not trivial to document in an easy to understand manner. Apart from these files, another way to get started is to compile the firmware, once your environment is set up, so you can test if any changes still compile (even if you don't have a supported keyboard, testing compilation is still a good idea).