keyboardio/Kaleidoscope #215: Set up the wiki to require pull requests

Jennigma, 2017-10-17:
I understand from this page that it's possible to have edits to the wiki require pull requests:

https://labs.inn.org/2014/05/19/applying-git-to-github-wikis/

This is the user facing documentation for using this system:

https://github.com/INN/largo/wiki/Contributing-to-this-wiki#contributing

I don't understand what is required to set it up, but it seems like a good system. I am open to thoughts, of course, and I can revert changes in the current system, but would like a little control. So far I've been catching and reviewing the changes people are making pretty quickly, but I would like to know changes don't go live without a second pair of eyes on them. I assume the time will come when I'm not in the repo several times a day, and at that point it will be nice to get a notice that I have something to review.

obra, 2017-10-17:
What that describes is basically not a wiki. It's online documentation rendered with Markdown. That's a great thing to have, but is a very different thing than a wiki. By definition, a Wiki ought to let pretty much any  community member contribute and update content on the fly.

The barrier imposed by https://github.com/INN/largo/wiki/Contributing-to-this-wiki#contributing is dramatically higher than "Click an edit button" -- In my experience, "drive-by" wiki contributions by previously-unknown-to-us parties have been both a huge source of valuable content and an amazing "hook" to draw in new contributors.

It might be the case that there are two classes of documentation and that some of it should be versioned next to the code with a limited set of contributors who can accept changes. I can totally believe that.

If there's an active spam or abuse problem, that is...very much a different story.

Jennigma, 2017-10-17:
ok. I'll keep an eye on it, and if there is an issue we can revisit. :-)

closed 2017-10-17 by Jennigma