keyboardio/Kaleidoscope #217: Investigate feasibility of turning off LEDS when the host goes to sleep or is turned off.

obra, 2017-10-21:


Jennigma, 2017-10-21:
I would like this feature. :-)

flackenstein, 2017-10-22:
@obra - I've been coding for the better part of 30 years and my C skills are a bit rusty.  Would really like to learn more about customizing the keyboard, so I guess there is no better time than the present.  Along with shutting down the LEDs I want to explore adding a fn + LED modifier to control led intensity.  Any info / suggestions on where to start would be greatly appreciated.

Thanks.

obra, 2017-10-22:
@flackenstein - Welcome!

The two tasks are going to likely touch very different parts of the code.

for the modifier to change LED intensity, there are two ways we might go about it. One is to modify the LED data as we send it out to the LEDs. Another would be introducing a "brightness" control into https://github.com/keyboardio/Kaleidoscope-LEDControl/blob/master/src/Kaleidoscope-LEDControl.cpp and modifying the stored data, though since there's code that reads that data and then performs updates based on it.

For turning off the LEDs when the host is off or suspended, you'd probably want to be looking at how the arduino core handles USBSuspend -- https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/cores/arduino/USBCore.cpp is where you'd start. At the same time, I'd probably want to make sure that it's still possible to have the LEDs turn on when the keyboard' getting power but no data, since that's something that some of us want.

flackenstein, 2017-10-25:
@obra - steaming ahead with the Kaleidoscope-SleepMode module and was hoping I could get a copy of the schematics?  Would like to know more on how the USB interface is setup and to see if I can use both the USB suspend and data inactivity to initiate the power-down mode and subsequently also set the LED mode to off.   In the power down state the Arduino loop code will be suspended until an external trigger (which I'm currently trying to figure and hence the need for schematics) will be used to resume loop execution.  I have most of the pieces figured out, just need a little more info to keep this task moving forward.    Thank you for the reset button, that has save my butt many times now :0)

obra, 2017-10-25:
Yup!  We've been meaning to publish 'em, but here's a quick copy of the
left side schematic, which should be everythign you need.

http://fsck.com/~jesse/tmp/2017-10-24/54c0ae8f-9664-484b-9358-8acaaee451dd/model01-left-hand.pdf

obra, 2017-10-25:
Oh, Also, if you hold in the "Prog" button when you connect your keyboard
to the computer, it will hang out in the bootloader. That should avoid
having to hit the built in RESET button.

Jennigma, 2017-10-25:
@obra fsck.com ? Cute. :-) Must have grabbed that early. One of my best friends has rtfm.com .

flackenstein, 2017-10-25:
@obra -- thanks for the schematics.   That fills in all the blanks.

algernon, 2017-11-25:
I did a bit of research about this, and with a small patch to Arduino core, this is possible to do. See my findings [here](https://github.com/algernon/keyboardio-worklog/blob/master/worklog.md#suspendwakeup-research). Will need to figure out how to proceed from here, but that's best done once we figured out why keypresses don't wake the host (#237).

algernon, 2017-12-09:
Reopening the issue until the `MyOldFriend` plugin is in Arduino-Boards and possibly in Model01-Firmware too.

closed 2017-12-16 by obra

cxw42, 2021-02-10:
@obra Thanks for the left-side schematic!  Would you please also post the right-side schematic?  I swapped out my keyswitches and have a few dead keys on the right that I'm trying to debug.  I have looked online and on the forums but can't find the schematic.  Thanks!