keyboardio/Kaleidoscope #218: "make flash" can't find files

MarkSenn, 2017-10-23:
I'm trying to install Kaleidoscope on a Fedora 26 Linux system using the following commands.  When I type "make flash" I get the following errors.  What am I doing wrong.  Thanks.

[Model01-instructions.txt](https://github.com/keyboardio/Kaleidoscope/files/1405563/Model01-instructions.txt)

obra, 2017-10-23:
Hi Mark,

We've actually got a much simpler install process we've been trialing.  It's written up here: https://community.keyboard.io/t/feedback-requested-simplified-software-installation/348?u=jesse

If you're game, I'd love to know if this solves the issues you're seeing. I'm sorry it's not the default in the README yet. 

Also, while we're happy to help you get things sorted out here, https://community.keyboard.io is a litltle better set up for support-like things, while GitHub is better set up for code-like things. (This is, of course, something you could not have known, because, well, I don't think I'd actually written those words down. I'm really sorry about that :/)

-jesse

jwakely, 2018-02-15:
I found this worked on Fedora 26 using the `arduino` packages from https://copr.fedorainfracloud.org/coprs/spot/arduino-1.8

```
su -c 'mkdir /usr/share/arduino/tools-builder /usr/share/arduino/hardware/tools'
# [enter root password]

AVRDUDE=avrdude \
AVRDUDE_CONF=/etc/avrdude/avrdude.conf \
ARDUINO_PATH=/usr/share/arduino \
ARDUINO_BUILDER=arduino-builder \
make flash
```

The makefile and scripts allow most things to be overridden with environment variables.

ryansb, 2018-03-05:
I ran into the same issue on Fedora, and it's the result of using a wildcard incorrectly in one of the variable discovery commands. Try this patch https://github.com/keyboardio/Model01-Firmware/pull/45 and see if it helps you.

Also, it seems odd that the AVRDUDE and other tool paths don't use commands that are in your path - I get having an override variable if it's unable to find the binaries, but I shouldn't have to specify all these `AVR_SIZE`, `AVRDUDE`, and so on if my path already has them. Is there a reason for avoiding checking the path that I'm not aware of?

jwakely, 2018-03-06:
@ryansb I guess my problem was different because that patch doesn't change anything for me. I don't have a `~/.arduino15/packages` directory so the command fails with or without the wildcard (and doesn't the `2>/dev/null` mean there's no error printed anyway?)

ryansb, 2018-03-07:
Also, those instructions (candidate new installs via add board in Arduino IDE) work great for me on Fedora 27. Thank you for linking those!

obra, 2020-06-08:
Closing this obsolete issue. Feel free to reopen if it's still an issue.,

closed 2020-06-08 by obra