keyboardio/Kaleidoscope #219: Feature request: a capslock LED mode that works analogous to the numlock LED mode

Soares, 2017-10-26:
See also https://community.keyboard.io/t/has-anyone-already-written-a-capslock-led-mode-analogous-to-the-numlock-led-mode/555

lydell, 2017-11-11:
A similar request: I’ve made a custom base layer. I use the Prog key to switch to the standard QWERTY layer, by setting the Prog key to `LockLayer(QWERTY)` in my custom layer and to `UnlockLayer(QWERTY)` in the QWERTY layer. I’d like the Prog key to be lit in blue when I’ve switched to QWERTY mode.

algernon, 2017-11-17:
@lydell That's reasonably easy to accomplish, and does not need a new plugin, with something like the following:

```c++
void QWERTYLEDHook(bool post_clear) {
  if (post_clear)
    return;

  if (Layer.isOn(QWERTY))
    LEDControl.SetCrgbAt(0, 0, CRGB(0, 0, 255));
  else
    LEDControl.refreshAt(0, 0);
}

void setup() {
  // ...
  Kaleidoscope.useLoopHook(QWERTYLEDHook);
}
```

lydell, 2017-11-18:
@algernon Thanks! That’s exactly what I needed! Just one small typo in there; it’s supposed to be `setCrgbAt` with a lowercase “s”. Sorry for hijacking the thread!

Soares, 2018-01-04:
I see this issue has been closed. Hooray! @algernon, would you mind giving me a brief summary of the right way to make caps-lock color the letters (but not the numbers) when activated, in analogy with numlock?

algernon, 2018-01-09:
@Soares Oops, closing was a mistake, as I did not explain the original question. Sorry!

I suppose the easiest would be to iterate over the keymap, find any keycodes between `Key_A` and `Key_Z`, and color those and those only. Something along these lines:

```c++
bool capsState = !!(kaleidoscope::hid::getKeyboardLEDs() & LED_CAPS_LOCK);

if (capsState) {
  for (uint8_t r = 0; r < ROWS; r++) {   
    for (uint8_t c = 0; c < COLS; c++) {
      Key k = Layer.lookupOnActiveLayer(r, c);

       if ((k.raw >= Key_A.raw) && (k.raw <= Key_Z.raw)) {
          LEDControl.setCrgbAt(r, c, CRGB(0, 0, 255));
        } else {
          LEDControl.refreshAt(r, c);
        }
    }
  }
}
```

Add this to your `loop` method, and it should do the trick.

closed 2018-01-09 by Soares

donfeduardo, 2018-02-12:
I've written a plugin with this functionality: [Kaleidoscope-CapsLock](https://github.com/donfeduardo/Kaleidoscope-CapsLock).