keyboardio/Kaleidoscope #220: Boot Protocol mode and negotiation is nor implemented.

obra, 2017-10-26:


algernon, 2017-10-26:
(related: https://github.com/keyboardio/KeyboardioHID/issues/10)

eltang, 2017-11-02:
Unfortunately, there isn't a straightforward solution to this problem. All BIOSes are supposed to indicate to a keyboard through a Set_Protocol request which protocol to use, so in theory all you would have to do is write a handler for this request which switches between the protocols. Unfortunately, things are not so simple in reality. A lot of BIOSes simply assume that all keyboards use boot protocol and skip this step. This doesn't cause problems for the vast majority of keyboards, since they always use boot protocol no matter what. Obviously this is not the case for Kaleidoscope-powered keyboards.

algernon, 2017-11-06:
We could do something similar to what QMK does: allow a manual switch. Hold some magic keys during keyboard boot, and it will go into 6KRO mode (or default to that, and the combo puts the keyboard into NKRO?).

While this is complicated too, and likely has a few corner cases, it sounds simpler than trying to negotiate a protocol, especially in light of what @eltang wrote above.

eltang, 2017-11-08:
Well actually there is one way to make Kaleidoscope magically work with BIOSes, but it's incredibly hacky, wastes endpoints, and might not be possible in Arduino. There's a good description of it [here](https://deskthority.net/workshop-f7/soarer-desparately-needed-t9322-30.html#p210882).

obra, 2017-11-08:
Thanks for the pointer, Eric.
Other things we need to do are going to require us to hack up Arduino's
PluggableUSB implementation, so that's not much of a worry.

As of right now, we have spare endpoints on the 32U4.

Likely we're going to have to do a bunch of real-world testing with the
various approaches. I know we'll want _some_ manual way to get the keyboard
into  BootProtocol mode, but I'd certainly prefer that, at least in the
more common situations, the end user not need to know that.
ᐧ

closed 2018-01-15 by obra

thatjpk, 2018-02-06:
Quick question: This appears to have landed after the Model01-Firmware v1.22 release was cut, so it's not in that version, but likely in the next?

tjkirch, 2018-02-25:
Related quick question: if you pull from git, is there a way to get this to work today?  I built latest, and saw these commits, but my BIOS doesn't recognize the keyboard.

I saw algernon post on the forums about setting "BootKeyboard.protocol = HID_BOOT_PROTOCOL" which I'm assuming disables NKRO entirely - is that the best way?  I don't really care about NKRO, or mouse support.

algernon, 2018-02-26:
> Related quick question: if you pull from git, is there a way to get this to work today? I built latest, and saw these commits, but my BIOS doesn't recognize the keyboard.

It should, once you set the default protocol, yes.

> I saw algernon post on the forums about setting "BootKeyboard.protocol = HID_BOOT_PROTOCOL" which I'm assuming disables NKRO entirely - is that the best way? I don't really care about NKRO, or mouse support.

No, it does not completely disable NKRO, it just makes the keyboard start in boot protocol mode (contrary to the USB spec, which says it should start in report - NKRO - mode). Once the OS tells the keyboard to switch to NKRO, it will. Windows, OSX, and Linux all do this, so setting the default to boot should be safe.

I'm tempted to suggest making boot the default anyway, precisely to avoid these kind of issues, even if we are then not 100% USB spec compliant. Because in practice, neither are BIOSes and OSes. :|

tjkirch, 2018-02-26:
I had to use "BootKeyboard.setProtocol(HID_BOOT_PROTOCOL);" instead, because 'protocol' is protected, so the simple assignment above didn't work.  I put that at the top of setup() and reflashed, but it didn't have any effect on my computer; I wasn't able to control the BIOS except with a backup keyboard.

Here's what I did - maybe there's some obvious mistake?  https://github.com/tjkirch/Model01-Firmware/commit/cc1240ab2b47dda2597743f1edea43d8f6694112

tjkirch, 2018-05-13:
I have to admit, this is one of my few frustrations with my Model 01, since it requires me to keep a spare keyboard around to control the bootloader menu after a reboot.  I keep trying the latest updates, but no luck with boot protocol so far.

Is there something that still needs to be enabled in terms of the default settings or merged into the mainline code?  Is there something that users need to enable in their config?  You can see the configuration I tried in the last comment.

tjkirch, 2018-05-16:
That didn't work.  Not only was the keyboard unresponsive to the BIOS/bootloader, it was unresponsive in Linux as well.  No keys were recognized, even after reboot.  :cry:

algernon, 2018-05-16:
Ow. I guess I have no choice but to dig deeper :|