keyboardio/Kaleidoscope #223: Tracking modifiers, and making isModifierActive more reliable

algernon, 2017-10-28:
Right now we are facing a problem where `isModifierActive` is not reliable, because it depends on the scanning order. Because we clear the report each cycle, when a key - a macro, or additional functionality provided by a plugin - wants to see the modifier state, and the key is reached before the held modifier, the report will not have the modifier pressed yet, and will erroneously report that the modifier is inactive. This renders `isModifierActive` rather useless.

I can see two reasonable solutions, one easy, but perhaps with more side effects, another a bit more complicated.

The first option would be to check not the current, but the previous report. Modifiers are usually held for more than a single cycle (even oneshots), so this is perhaps a reasonable compromise. The advantage of this method is that it is easy, and will be correct much more often than what we have now. So perhaps, it would make sense to do this change until we figure out something better.

As for something better... what if we had a bitmap of all the keys, and could mark keys as needing special attention? We could process these first, before anything else. Modifiers could mark themselves special on press, and clear it on release. This way `isModifierActive` would always be correct. Allowing plugins to mark their triggers special is something I feel could be useful, but I'm not entirely sure about it.

obra, 2017-10-28:
The problem statement makes sense. I think all my personal use cases for
isModifierActive were "post-scan"

I don't think that marking modifier locations is going to be 'enough',
because macros and plugins can modify the state.

Checking modifiers from the previous scan, in theory, sounds plausibleish.
Though I think that what we'd want to do is to add an API to steal the
modifier state, or possibly the whole keymap state, from the previous cycle
and expose an API to it.

algernon, 2017-10-28:
Exposing an API to work with the previous state makes sense... but then `isModifierActive` should likely become `wasModifierActive`, and check the previous state. Because in its current form, it is very misleading, as it turns out. :(

obra, 2017-10-28:
It’s accurate....depending on when you call it. I think.

gedankenexperimenter, 2017-11-23:
How about making `isModifierActive()` call `wasModifierActive()` if it's called in the `eventHandlerHook` pass & post-clear `loopHook`, but give the current state in pre-clear `loopHook`s?

That way, plugins that really care about precisely what goes into each individual report can still get good information about the modifiers that will be in the next report, and ones that don't need to worry about precise timing won't be mislead by the name and get blatantly false information. To make it even better, there could be a repeat of the pre-clear `loopHook` pass if the modifier state changed during the first pass.

algernon, 2017-11-23:
I'm not sure if `isModifierActive` calling `wasModifierActive` in certain cases would make things *less* confusing (and it would certainly complicate the code). :/

Running the pre-clear `loopHook` twice a cycle, but only sometimes is a kind of uncertainty that is very hard to debug. Not to mention it has the potential of adding considerable latency.

gedankenexperimenter, 2017-11-23:
> I'm not sure if isModifierActive calling wasModifierActive in certain cases would make things less confusing (and it would certainly complicate the code). :/

Is it worse than `isModifierActive()` returning the wrong answer in most situations?

algernon, 2017-11-23:
Even if not worse, it is much more complicated to achieve, and the gains aren't that big to warrant the effort. I mean, one can use `wasModifierActive()` right now, in every case you outlined above that would make `isModifierActive` equivalent. To make `isModifierActive` act like you described, we'd have to store the current state somewhere, which is both more code, more data, and more complex.

Better documentation for both functions would help more, if you ask me.

gedankenexperimenter, 2017-11-23:
I agree that better documentation would help, but it would still be so easy to make the mistake of testing by calling `isModifierActive()` in an `eventHandlerHook`, where you're very likely to get a false negative. Even with good documentation, having a function named `isModifierActive` that only returns the correct answer (somewhat) reliably during one out of three hook passes is just begging for plugin bugs.

gedankenexperimenter, 2017-11-23:
Looking at the code more, I'm leaning towards agreeing with you about not trying to hack `isModifierActive` to make it more reliable (but still not perfectly reliable). There is one other idea I have that I think might make plugin developers less likely to run into this problem, though: maybe changing the names of the hooks would help.  Right now we have three, two of which have the same very generic name (`loopHook`), and one of which suggests (to me, anyway) that it's where all the real work should get done (`eventHandlerHook`).  Maybe these could have names that better describe when they run: `scanHook`, `postScanHook`, and `postClearHook`, perhaps? Or `interScanHook`/`postScanHook`/`postSendHook`? Or `duringScanHook`/`preReportHook`/`postReportHook`?

This idea would deprecate some much-used interface names, so if y'all think it's a good idea, it might make sense to combine it with #240.

obra, 2017-11-23:
Changing the names of the hooks is something I'm only willing to do in a major version upgrade. 

Before we consider any renaming, we should exhaustively document the existing hook points (with comments in code, describing the calling conventions, the behavior, the state expected, the state guaranteed and the state not guaranteed for each hook. 

Adding additional hook points is something I'm a little more comfortable doing 'soon', though adding a new name for the same hook point or two hooks that run 'next to' each other is something that I believe will lead to more confusion, rather than less confusion.

I do think that there are places we can add hooks that will give plugin authors more rope to play with, but want to set a pretty high bar for adding hooks, so that we don't run into a situation (again) where we're boxed in by the implementation. For new hooks, I'd love to see specific use cases that the hook enables that couldn't be done before.

Additionally, as @algernon mentioned, we need to be really careful about the performance overhead of new hook points. Unless #240 or something like it addresses it, adding a hook adds O(plugins) function calls, which, on our tiny embedded processor..actually matter for typing latency.

gedankenexperimenter, 2017-11-23:
> Changing the names of the hooks is something I'm only willing to do in a major version upgrade.

Certainly!

> Before we consider any renaming, we should exhaustively document the existing hook points (with comments in code, describing the calling conventions, the behavior, the state expected, the state guaranteed and the state not guaranteed for each hook.

I feel like I'm getting close to being able to do a lot of that documentation. The main thing I'm lacking before I really get started is guidelines on where you want that documentation, and the format to put it in.

obra, 2017-11-27:
In general, I'd like function-level documentation inline above the function
in question as a comment in a form that doxygen can extract. Everything
else, I'd probably push to the wiki.