keyboardio/Kaleidoscope #227: Consider how to make LockLayer() override ShiftToLayer()

gedankenexperimenter, 2017-10-31:
If a user defines a keymap entry as `LockLayer(N)` and another as `ShiftToLayer(N)`, even after layer `N` is locked, it will be deactivated by releasing the `ShiftToLayer(N)` key. This is not the way that CapsLock and Shift work on a standard keyboard, and might be better to change so that there's a simple way to define it such that a layer really does get "locked" just like CapsLock.

In fact, unless I misunderstand the code, if I define a layer `N` with a `ShiftToLayer(N)` key on it, pressing and releasing that key will turn off layer `N`. For a more real-world example see the [discussion on the forum](https://community.keyboard.io/t/multiple-layers/401/44?u=merlin). @algernon had some idea about implementing this feature with a plugin, and asked me to create this issue, but I admit that I did not understand his idea, so I'll let him explain.

obra, 2017-10-31:
> If a user defines a keymap entry as LockLayer(N) and another as
> ShiftToLayer(N), even after layer N is locked, it will be deactivated by
> releasing the ShiftToLayer(N) key. This is not the way that [...]

That's actually OS dependent, which makes the correct behavior even less
clear.

On macOS, Tapping CapsLock, then typing the letter 'u, then holding down
Shift and typing 'ppercase' generates = UPPERCASE
On Linux, and, I believe also on Windows, tapping CapsLock, then typing the
letter 'u, then holding down Shift and typing 'ppercase' generates =
Uppercase

(I just typed that on Linux to be 100% sure)

gedankenexperimenter, 2017-11-01:
Yes, but that's not what I meant. Is there any OS where tapping `CapsLock`, then pressing _and releasing_ `shift` turns off CapsLock?  Because if you tap `LockLayer(N)`, then a release of `ShiftToLayer(N)` will turn off layer N.

gedankenexperimenter, 2017-11-01:
In fact, it might be a useful feature to have `ShiftToLayer` also work as a toggle. If layer N is "locked" on, holding down `ShiftToLayer(N)` could temporarily turn layer N _off_ (à la `shift` & `CapsLock` on Linux).

closed 2018-01-25 by gedankenexperimenter