keyboardio/Kaleidoscope #229: Document Kaleidoscope workflow (i.e. working with many repositories)

gedankenexperimenter, 2017-11-01:
@obra mentioned that he has a good way of working with the repository structure (Arduino-Boards and its submodules). It would be useful to other prospective contributors to have procedures for managing the many repositories required to build and flash the firmware while working on their own changes to a subset of the modules. See [this forum topic](https://community.keyboard.io/t/working-with-multiple-repositories/651).

gedankenexperimenter, 2017-11-01:
I'm hoping for a process that allows me to check out `Arduino-Boards` and all of its submodules (e.g. `keyboardio/Model01-Firmware`), and override them individually from a different path with just the modules that I've forked for working on (e.g. `gedankenexperimenter/Model01-Firmware`), without needing to fork `Arduino-Boards` itself and rewrite `.git-modules`.

Maybe setting up the makefile to read an environment variable so it can override the include path would work?

gedankenexperimenter, 2017-11-16:
I'm thinking of writing something up that would have procedures for most of the things a contributor would want/need to do. @obra, I'm thinking of describing things in a Kaleidoscope wiki page — would that suit you, or would you rather have it go elsewhere (e.g. Arduino-Boards)?

obra, 2017-11-16:
A Kaleidoscope wiki page is great. Thank you!
ᐧ

gedankenexperimenter, 2017-11-16:
Also, it has occurred to me in my puttering around with the code that it would be helpful (to me, anyway) if there were rules and/or guidelines for naming branches (at least those used to submit pull requests). It's…inconvenient if two contributors use different conventions, and could be a bit problematic if they use identical branch names in different forks of the same repository.

obra, 2017-11-16:
If a pull request is coming from a personal repo, the branch name really
doesn't matter. But for work inside the core repos, yes, that'd be great.
It probably belongs in the coding style guide.
ᐧ

Jennigma, 2017-11-16:
I might chunk it out into a couple pages; perhaps "setting up a dev environment," "developing in the Kaleidoscope dev environment," and "submitting pull requests?"

gedankenexperimenter, 2017-11-16:
> A Kaleidoscope wiki page is great. Thank you!

I probably won't get much done on it until Monday, unfortunately; I'm going to have kids home from school for the rest of this week.

gedankenexperimenter, 2017-11-16:
> I might chunk it out into a couple pages; perhaps "setting up a dev environment," "developing in the Kaleidoscope dev environment," and "submitting pull requests?"

That sounds fine to me. I'll start with an overview, and probably we'll spin off sections into separate pages as it makes sense to do so.

Jennigma, 2017-11-16:
Just let me know how I' can help. Thanks for this!

benshine, 2017-12-04:
FWIW I got a stub page started at [Developer Setup](https://github.com/keyboardio/Kaleidoscope/wiki/Developer-Setup) with what I know so far. It's a prereq for actually contributing, although not sufficient instructions for contributing.

closed 2025-06-01 by obra