keyboardio/Kaleidoscope #232: macOS: stty: /dev/cu.usbmodemCkbio011: Device not configured

Jennigma, 2017-11-02:
Opening at @obra's request in response to thread here:

https://community.keyboard.io/t/macos-stty-dev-cu-usbmodemckbio011-device-not-configured/493

Initial message from: ackintosh (Akihito Nakano):

> I’ve tried to change key layout. Displayed the following error.
>
>Arduino IDE 1.8.5
>$ pwd
>/Users/akihito1/Documents/Arduino/
>$ ls
>Model01-Firmware hardware
>$ cd Model01-Firmware
>$ make flash
>BOARD_HARDWARE_PATH="/Users/akihito1/Documents/Arduino//hardware" >/Users/akihito1/Documents/Arduino//hardware/keyboardio/avr/libraries/Kaleidoscope/bin//kaleidoscope-builder flash
>Building output/Model01-Firmware/Model01-Firmware (0.0.0-gv1.13-55-g8ab1) ...
>Press ENTER when ready...
>
>stty: /dev/cu.usbmodemCkbio011: Device not configured
>make: *** [flash] Error 1
>the keyboard is connected
>LED works, key-input works
>Device not configured … :confused:
>
>Any suggestions for how to debug?
>
>Thank you.

Second response from: blakej (Blake Jones)

>I just got a swap of the circuitry of the left hand of my keyboard, and now I’m seeing this as well. (My error message is has “kbio01”, rather than “kbio011”, but otherwise it’s identical.) I’ve tried holding down PROG while pressing Enter; holding down PROG before typing “make flash” and keeping it held until after the “Press ENTER” prompt; unplugging the keyboard, starting the “make flash”, then holding down PROG while plugging it in and pressing ENTER at the prompt. All give the same failure.
>
>I think I saw this once or twice before the circuitry swap, but eventually I did something slightly different (sadly I don’t remember what) and it worked. But now I’m getting no luck.
>
>I haven’t tried the IDE yet; even if that does work, it seems like the CLI issue should be understood better.

adkron, 2018-01-31:
I have the exact same bug. It works from the IDE.

lfurr, 2018-03-10:
I'm seeing this now too, after successfully flashing my keyboard several times from the command line (including twice earlier today). The only thing I'm aware of having changed since the last successful flash was the USB and Cat5e cables, but I changed the cables back and I'm still seeing this.

stty: /dev/cu.usbmodemCDkbio01: Device not configured
make: *** [flash] Error 1

Process:
1) Edit the Model01-Firmware.ino file in the Arduino IDE.
2) In Terminal on my Mac, go to ~/Documents/Arduino/Model01-Firmware and run the command `make flash`.
3. When prompted, hold the Prog key and press Enter.

obra, 2018-03-10:
>    1. When prompted, hold the Prog key and press Enter.

Just to confirm, it still works correctly from the Arduino IDE? If so,
there's something wrong with the new port detection logic.

lfurr, 2018-03-10:
Nope, I just edited my comment to say I never have been able to flash from the IDE, so I can't use that for diagnosis.

lfurr, 2018-03-10:
Currently in the IDE I get
Arduino: 1.8.5 (Mac OS X), Board: "Keyboardio Model 01"

Sketch uses 20658 bytes (72%) of program storage space. Maximum is 28672 bytes.
Global variables use 2040 bytes (79%) of dynamic memory, leaving 520 bytes for local variables. Maximum is 2560 bytes.
Low memory available, stability problems may occur.
Couldn't find a Board on the selected port. Check that you have the correct port selected.  If it is correct, try pressing the board's reset button after initiating the upload.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

obra, 2018-03-10:
If you go to the Port drop down, do you see any options?

lfurr, 2018-03-10:
At the moment, half the keyboard is in pieces on my lap--I thought I'd try looking for the reset button. (Speaking of which, I was surprised not to find any documentation saying where the reset button is. My search-fu may be at fault.) As soon as I finish reassembling I'll check and reply.

obra, 2018-03-10:
Alternatively, if you type the following at a command line prompt:

ls -al /dev/cu.usbmodem*; sleep 10; ls -al /dev/cu.usbmodem*

and then, while holding down the 'Prog' key, very quickly disconnect your
Model 01 from your computer and plug it back in, what output do you get?

obra, 2018-03-10:
With the current bootloader, you should pretty much never need the hardware
reset button. The 'Prog' key puts the keyboard into bootloader mode when
held in at boot, just as the reset button would.

https://community.keyboard.io/t/how-the-prog-key-gets-you-into-the-bootloader/506/6?u=jesse
should get lifted to the wiki.

lfurr, 2018-03-10:
In the Arduino IDE, Tools -> Ports lists /dev/cu.Bluetooth-Incoming-Port and /dev/cu.usbmodem36 (Keyboardio Model 01)

and the command got me:
```
~/Documents/Arduino/Model01-Firmware$ ls -al /dev/cu.usbmodem*; sleep 10; ls -al /dev/cu.usbmodem*
crw-rw-rw-  1 root  wheel   21,  73 Mar  9 19:54 /dev/cu.usbmodem36
crw-rw-rw-  1 root  wheel   21,  55 Mar  9 18:38 /dev/cu.usbmodemCDkbio01
crw-rw-rw-  1 root  wheel   21,  55 Mar  9 18:38 /dev/cu.usbmodemCDkbio01
crw-rw-rw-  1 root  wheel   21,  75 Mar  9 19:57 /dev/cu.usbmodemkbio01
```

Thanks for the info about the reset button!

lfurr, 2018-03-10:
I can now flash in the IDE, so there's one good thing--neither port was selected, so I tried selecting one.

obra, 2018-03-10:
Ok. There's definitely something up.

It looks like you've got a phantom device entry (usbmodemCDkbio01) hanging
around confusing our too-simple search for ports.

The fact that your keyboard is coming up as 'usbmodem36' suggests that it
might have been most recently flashed with the IDE set to 'Arduino
Leonardo' or something else rather than Model 01. That's not actually a big
deal, but can cause the port name to not be our port name.

lfurr, 2018-03-10:
In the IDE, the board is set to Keyboardio Model 01.

When I looked again at the port, it was usbmodem40, and now usbmodem42. I have no idea what that change signifies.

lfurr, 2018-03-10:
I tend to leave the computer on through the work week, I wonder if a restart would clear out the phantom device entry. I'll give that a try.

lfurr, 2018-03-10:
Well, after the reboot everything is better! I'm sorry for troubling you with something that rebooting would fix, but I am glad that I understand the Model 01 even better now. `make flash` works, the IDE lists my ports as /dev/cu.Bluetooth-Incoming-Port and /dev/cu.usbmodemCDkbio01 (Keyboardio Model 01) and my devices are 
```
crw-rw-rw-  1 root  wheel   21,   7 Mar  9 20:25 /dev/cu.usbmodemCDkbio01
crw-rw-rw-  1 root  wheel   21,   9 Mar  9 20:28 /dev/cu.usbmodemkbio01
```

Thank you!

obra, 2018-03-10:
There's no need to apologize. We should _still_ be doing better
bulletproofing so nobody else gets bitten by this kind of hassle.

I am thrilled that it's now behaving, though :)

qayshp, 2018-05-26:
Just hit the same issue on macOS 10.13.4, and about to try the reboot method. Looks like I also have a phantom device as above from @lfurr.

```
// attached normally
crw-rw-rw-  1 root  wheel   21, 111 May 25 20:29 /dev/cu.usbmodemCDkbio01
crw-rw-rw-  1 root  wheel   21,  59 May 25 20:11 /dev/cu.usbmodemkbio01
// 10 seconds later, reattached while holding PROG
crw-rw-rw-  1 root  wheel   21, 113 May 25 20:29 /dev/cu.usbmodem56
crw-rw-rw-  1 root  wheel   21,  59 May 25 20:11 /dev/cu.usbmodemkbio01
// no keyboardio attached
crw-rw-rw-  1 root  wheel   21,  59 May 25 20:11 /dev/cu.usbmodemkbio01
```

qayshp, 2018-05-26:
Reboot got rid of the phantom device and now the keyboard was able to be flashed by the make target.

Kinda wondering what puts the mac in this state.

algernon, 2018-05-26:
#301 may be of help here - I merged that to master for easier testing. Can you try updating Kaleidoscope, and see if you can still reproduce the problem?

qayshp, 2018-05-31:
I've updated, and I haven't had this problem since. I also haven't seen the leftover ghost entry for `cu.usbmodemkbio01` in `/dev/` with nothing attached since then either.

gedankenexperimenter, 2019-05-06:
I think this has been resolved by #301.

algernon, 2019-05-10:
Lets mark it as such, then.

closed 2019-05-10 by algernon