keyboardio/Kaleidoscope #234: issues with IDE setup documentation

aspiers, 2017-11-03:
The "Install support for your keyboard into the IDE" section is duplicated between https://github.com/keyboardio/Kaleidoscope/wiki/Install-Arduino-support-on-Windows and https://github.com/keyboardio/Kaleidoscope/wiki/Install-Arduino-support-on-a-Mac, but missing from https://github.com/keyboardio/Kaleidoscope/wiki/Install-Arduino-support-on-Linux.

I suggest refactoring that duplication out into a new page and then having all 3 point to it.

Jennigma, 2017-11-05:
Once we settle the Win 7 issues for good, I will merge Mac and PC and have a separate page for the Linux zoo since there are so many varieties each with their own quirks. I think I can merge Mac and PC without it getting too complex for readers, but I despair of Linux.

At any rate, I know this will need a refactoring, but I have other things I want to get into at least first draft before I go back to revise. :-)

aspiers, 2017-11-12:
Sounds good! Thanks a lot :)

obra, 2020-06-08:
I believe this problem was resolve a while ago. Closing.

closed 2020-06-08 by obra