keyboardio/Kaleidoscope #236: Compiler warning and none functional 2bytes Consumer Keys

SjB, 2017-11-05:
When I insert the **Consumer_AL_Local_MachineBrowser** Key in my keymap I get the following compiler warning:

```
Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope/src/key_defs_consumerctl.h:190:139:` warning: narrowing conversion of '404' from 'int' to 'uint8_t {aka unsigned char}' inside { } [-Wnarrowing]
 #define Consumer_AL_Local_MachineBrowser (Key) { HID_CONSUMER_AL_LOCAL_MACHINE_BROWSER, KEY_FLAGS | SYNTHETIC|IS_CONSUMER  | HID_TYPE_SEL }
```

Sure it's just a warning but the function doesn't work since the Key.keycode is only an int8. So the second byte gets truncated and the key does not work. 

My current solution is to use the Macro and ConsumerControl plugin.

```c++
const macro_t *macroAction(uint8_t macroIndex, uint8_t keyState) {
  switch (macroIndex) {

  case MACRO_VERSION_INFO:
	  versionInfoMacro(keyState);
	  break;
  case MACRO_FILES:
	  if (keyToggledOn(keyState)) {
		  ConsumerControl.write(HID_CONSUMER_AL_LOCAL_MACHINE_BROWSER);
		  Keyboard.sendReport();
	  }
	  break;
  }
  return MACRO_NONE;
}
```

algernon, 2017-11-05:
For the time being, a macro is the best solution indeed. In the long run, we'll likely want to adjust these defines so that if `IS_CONSUMER` is set, we'll add the second byte in the event handler, via a switch statement or something along those lines.

SjB, 2017-11-30:
This bug was fixed with https://github.com/keyboardio/Kaleidoscope/pull/242 pull request

closed 2017-11-30 by SjB