keyboardio/Kaleidoscope #237: Keyboard doesn't respond to keypresses when Windows session is locked

kajsa, 2017-11-05:
When I come back to my desktop machine after a time away (long enough that Windows puts it to sleep), none of the buttons on the Model 01 register keypresses unless I unplug the keyboard and plug it back in.

I noticed this morning that the LEDs were still glowing, and still shifting through the rainbow cycle. To try to get some more information, I set the LED mode to Stalking, turned the sleep timeout down, and stepped away. Coming back, I can see based on the LED behavior that the board is processing keypresses, but not communicating them to the host.

wez, 2017-11-05:
FWIW, in my experiments with suspend/sleep modes I found that the arduino usb stack did not survive coming back from sleeping the CPU.  The LUFA implementation works just fine, but isn't an option for arduino.  I'd recommend avoiding putting the controller to sleep if possible.  I haven't been following the core of the kaleidoscope firmware very closely recently, so I don't know if this is something changed in the kaleidoscope code or something that may have changed in arduino itself.

kajsa, 2017-11-05:
Interesting. Since I'm dealing with a desktop, I can certainly set it not to go to sleep as a workaround.

obra, 2017-11-06:
Indeed, I spoke to a domain expert today and got similar advice. the amount of power we burn even without sleep is...small. simply defaulting to turning off the LEDs would be 90-98% of all the possible power management win.

algernon, 2017-12-09:
I got to the point that I can wake the host up after it suspended, under Linux. Needed to explicitly enable wakeup, it didn't enable automatically. I suppose that the same is possible under Windows, allowing wakeup on a per-device basis.

It would be better if we could Just Do That(tm) automatically, and am trying to figure out how to enable remote wakeup by default.

algernon, 2017-12-09:
The [MyOldFriend](https://github.com/keyboardio/Kaleidoscope-MyOldFriend) plugin has a fix for this, I hope. It allows the keyboard to wake the host up on Linux at least. I have not tested any other platforms yet, but have my fingers crossed.

algernon, 2018-01-30:
This has been fixed by keyboardio/Model01-Firmware#41.

closed 2018-01-30 by algernon