keyboardio/Kaleidoscope #239: Holding down multiple modifier keys on the right half often results in no key-up events for the modifiers

imrekoszo, 2017-11-06:
Using the stock firmware at https://github.com/keyboardio/Model01-Firmware-Builds/blob/master/v1.13-MP1-BUILD/atmega32u4_firmware.hex I have the following issue:

I use a few multi-stroke shortcuts, mainly in Intellij Idea, for example `Alt+Shift+T, Alt+Shift+R`. I execute that the following way:

1. Press and hold `alt` and `shift` on the right half
2. Press and release `T`
3. Press and release `R`
4. Release `alt` and `shift`

This often results in `LeftAlt` and/or `RightShift` to get stuck, as in my computer interpreting them as still being held down. If I continue typing it won’t do what I would expect it to do. To fix this I have to press the two modifier keys again and then it’s fine.

Further investigation details:
- This hasn’t happened using any other keyboard
- It doesn’t happen if I use the modifier keys separately, like Ctrl+C, Ctrl+V or holding down Shift to type uppercase text
- Could not reproduce it using the left half of the keyboard or by holding one modifier on the left and the other on the right
- Could not reproduce it using the buttons labeled shift and ctrl on the default arrangement on right half no matter what firmware I used
- Reproduced it using any 2-button combination of shift, alt and space on the default arrangement on the right half (using modified firmware)
- Reproduced it using any 2-button combination of alt, space and ctrl on the default arrangement on the right half (using modified firmware)
- The issue seems to be related to the physical buttons rather than which modifier is mapped to them. With custom firmware I can reproduce this using `Alt+Shift` and `Ctrl+Shift` too when using the physical button combinations in previous points

Please see original discussion here:
https://community.keyboard.io/t/holding-down-multiple-modifier-keys-on-the-right-half-often-results-in-no-key-up-events-for-the-modifiers/673

algernon, 2017-11-24:
I'm having trouble reproducing this, using current master of... pretty much everything. As it appears to be physical location related, I have the feeling this may not be a firmware issue, but a hardware one.

If you move the `Alt` and `Shift` keys on the right to somewhere else, can you still reproduce the problem? Or, if you put, say, `A` and `S` on these physical keys, do they remain stuck too, from time to time?

imrekoszo, 2017-11-24:
It could very well be a hw issue. I recorded a short video demonstrating the problem: https://youtu.be/rapOiyf1OFY

Notes:
- I remapped the right thumb cluster to `U I O P`
- I managed to repro the sticking by just pressing pairs of keys in the right thumb cluster
- Contrary to my earlier tests, using any 2 keys from the right cluster at the same time can achieve this
- When keys get stuck, I can unstuck them by pressing any of the keys on the right half, even the likes of `fn`. Pressing keys on the left half does not achieve this (this is not in the recording therefore)
- The issue seems to only affect the 4 thumb keys. I haven't yet achieved this using any other keys.

imrekoszo, 2018-01-08:
Minor update: I have reproduced this using different ethernet cables connecting the two halves, this was something I hadn't tried before.

obra, 2018-01-08:
I have a new key scanner firmware I’m almost ready to hand you to test. I’m so sorry for the delay.

imrekoszo, 2018-01-08:
No worries, I'm happily using the left half for double modifiers in the meantime.

obra, 2018-01-16:
Oh hey. I've managed to actually reproduce this issue. It's definitely (at
least in my case) firmware and not hardware.

imrekoszo, 2018-01-16:
Fantastic news! I was secretly hoping it's a sw problem. Did you repro it with the key scanner fw or the normal one? Let me know if you'd like me to try something. (although I'm separated from my model01 for a few more days)

obra, 2018-01-18:
It's going to be a little longer before I have something I really trust,
but I have something that an advanced user could try whenever you say the
word.

It's a new keyscanner firmware, to be updated by flashing a special
firmware to your Model 01

imrekoszo, 2018-01-18:
I'm happy to try it if you think it could be helpful in resolving the issue or even getting closer to understanding it. Not sure how advanced of a user you're referring to; I maintain my own fork of model01-firmware and flashed a number of different versions to my keyboard but I haven't really explored or understand the internals of Kaleidoscope or its plugins.

closed 2020-06-08 by obra