keyboardio/Kaleidoscope #243: Use of Key_KeymapNext_Momentary could result in reading past the end of the keymap array

gedankenexperimenter, 2017-11-10:
In `layers.cpp`, the function [`Layer.next()`](https://github.com/keyboardio/Kaleidoscope/blob/3f1c80366a310438ee9bef50dff3eb0dd394ac85/src/layers.cpp#L157) can result in reading off the end of the [`keymaps[]`](https://github.com/keyboardio/Model01-Firmware/blob/8f5851138cd3b22ed583ec91f82be6f449af4562/Model01-Firmware.ino#L127) array if `Key_KeymapNext_Momentary` is used on the highest defined layer. Right now, there's no way to check and guard against this, because the total number of layers defined in that array is not accessible from `layers.cpp`.  This also potentially affects calls to `Layer.on(N)`, but that's more difficult for a naïve user to trigger.

This could be addressed by adding a line to the sketch which makes the number of defined layers accessible. For example, the following line in `Model01-Firmware.ino` ought to enable such a check:

    const uint8_t DefinedLayersCount PROGMEM = sizeof(keymaps) / sizeof(*keymaps);

gedankenexperimenter, 2017-11-10:
Incidentally, exposing the number of defined layers would enable us to make other things somewhat more efficient, too. See [this discussion](https://community.keyboard.io/t/how-to-get-the-number-of-defined-layers/647?u=merlin) on the forum.

gedankenexperimenter, 2017-11-10:
I'd be happy to submit a PR for this, but since it requires (as far as I can tell) changes to more than one repository, I'm not sure how to go about doing that (if the change is made only in Kaleidoscope, it would break, so Model01-Firmware would need to be first). I'm also not sure @obra & @algernon would like this solution, which seems a bit fragile.

gedankenexperimenter, 2017-11-29:
I think this can be closed now, though it's only really fixed if the `KEYMAPS()` macro is used (or `layer_count` is otherwise set correctly) in the sketch.

algernon, 2017-11-29:
As #244 is the best we can do while maintaining backwards compat, I agree, this can be closed now.

closed 2017-11-29 by algernon