keyboardio/Kaleidoscope #245: kaleidoscope-builder might be able to detect if arduino-boards was checked out from git and hasn't had its submodules syncced

obra, 2017-11-14:


gedankenexperimenter, 2017-11-18:
`git submodule foreach` should be helpful for this, with `git remote update`, then either `git status --porcelain -uno` or some clever `rev-parse` and `merge-base` comparisons. I'm not sure it can be done without a `git fetch` for the remote in question, but that shouldn't be a real problem, since it doesn't require a `git merge`. One tricky bit is if the checked-out branch isn't `origin/master`, particularly if its remote isn't `keyboardio/<PLUGIN>`, or if it doesn't have a remote at all.

I'm guessing you just want it to print a warning so that when someone runs the build, they find out if they need to sync their local branches?

algernon, 2022-06-08:
We no longer use - or even have - kaleidoscope-builder, so I suppose this is obsolete?