keyboardio/Kaleidoscope #246: (Re)defining a ShiftToLayer(N) key on layer N causes that layer to not turn off

gedankenexperimenter, 2017-11-20:
If a key is defined on the base layer as `ShiftToLayer(N)`, and that same key is defined as something else on layer `N`, pressing and releasing that key will turn on layer `N`, but won't turn it off.

This happens because `updateLiveCompositeKeymap()` gets called on both key press and release events. It can be fixed by only updating on presses, instead. @algernon, @obra: Is there any reason that we need to update the keymap on release events?

algernon, 2017-11-20:
The reason we want to update on release too, is because of the following scenario:

- Press and hold a `ShiftToLayer` key.
- Press and hold a key that is modified by the target layer, and can repeat.
- Wait.
- Release the `ShiftToLayer` key, but continue holding the other one.
- Wait.
- Release the other key too.

Lets suppose we held the `h`/`left arrow` key along with `Fn`. If we do not update the cache on release, the key will stay `left arrow` until pressed again. While that won't affect its behaviour much, it will affect any other code that does something based on the keymap, such as highlighting arrow keys, or anything similar.

gedankenexperimenter, 2017-11-20:
> While that won't affect its behaviour much, it will affect any other code that does something based on the keymap, such as highlighting arrow keys, or anything similar.

What about keys that a plugin wants to hightlight, but which were never pressed? When do they get updated in the live keymap cache?

gedankenexperimenter, 2017-11-20:
So, for example, I press and hold an `Fn` key, and then want the arrow keys highlighted. But those keys haven't toggled on _or_ off, so they're still `h`, `j`, `k`, and `l` in the `liveCompositeKeymap[]`, aren't they? Updating it on release as well as press doesn't provide the benefit you described. The only other place I can find `updateLiveCompositeKeymap()` being called is in `Kaleidoscop.setup()`.

algernon, 2017-11-20:
> What about keys that a plugin wants to hightlight, but which were never pressed? When do they get updated in the live keymap cache?

The plugin can use the `lookupOnActiveLayer` there. On second thought, LED effects should probably use that to begin with.

So I went looking for other places where we use `Layer.lookup`, and only found a use in `DualUse` outside of Kaleidoscope core (and in `-LEDEffects`, but that should probably be using `lookupOnActiveLayer` instead).

Which means I no longer have a good argument against the change, it seems.

PR #247 (2017-11-26): Only call updateLiveCompositeKeymap() on key press; not release
If we call updateLiveCompositeKeymap() on key release the keymap gets
updated before the release event occurs, and any ShiftToLayer(N) key
with a different definition on layer N won't work properly. Before its
release event is processed, it gets updated to the new value, and
layer N doesn't get turned off. If we only update the live keymap on
key press events, we don't have this problem.

Fixes #246
closed 2017-11-26 by obra