keyboardio/Kaleidoscope #249: Should Layer::on() be renamed to make it clear that it's actually for turning on the layer?

obra, 2017-11-23:
See PR #230.

"Amusingly, I changed it fro isOn() to directly using bitRead() because I thought it was clearer (because of the next line, which calls bitSet()/bitClear()), not because I thought the function call would incur any overhead. I can see your point of view, though."

That led me to be about to suggest that we should be adding a 'Layer::turnOn' function instead of the bitSet...until I realized that 'on' is supposed to be that function... It's not a blocker for merging this, but I wonder if we should be renaming 'Layer::on to Layer::turnOn and adding a deprecation warning for Layer::on. I'll open an issue for discussion.
"

gedankenexperimenter, 2017-11-23:
Interesting. My point of view was that we were already looking below the abstraction barrier that `Layer.isOn()` was providing, so it was actually more of a burden to me when trying to understand the code to call it instead of `bitRead()`; I had to search in two places to figure out what was going on, rather than just one. It never occurred to me that `Layer.on()` could be interpreted as a test rather than a setter, but now that you point it out, it seems like it could cause confusion.

gedankenexperimenter, 2017-11-23:
If you do want to verbify the function name, I'd suggest `activate()` rather than `turnOn()`, which would be consistent with the `activeLayers[]` array and the `updateActiveLayers()` function (and I think there are a bunch of other places where some form of the term "active" is used). That would lead naturally to a few other name changes:

- `Layer.activate()`
- `Layer.deactivate()`
- `Layer.isActive()`
- `Layer.topActiveLayer()`
- `Layer.updateTopActiveLayer()`

…and possibly a few more.

algernon, 2017-11-23:
I like the "active"/"activate" based scheme.

obra, 2017-11-27:
Sure. That works for me. Leaving the other names in place for now with
deprecation warnings isn't wrong.

algernon, 2017-11-27:
@gedankenexperimenter wanna go ahead with the renames, or shall I?

gedankenexperimenter, 2017-11-28:
I'll go ahead and do it if you like, but I've been busy with unrelated things today, so I might not get around to it until tomorrow or so.

algernon, 2017-11-28:
That's fine, it's not urgent. (and I'm off to bed, too)

gedankenexperimenter, 2017-11-30:
I'm going to wait until the pending changes to `layer.*` are merged before doing this rename in order to avoid merge conflicts.

gedankenexperimenter, 2017-12-01:
…and now it's done, but I probably won't have time to work on Kaleidoscope again until Monday.

algernon, 2017-12-01:
I'll go ahead and do it then, if I have a few spare cycles.

PR #260 (2018-12-10): Layer: Rename a few functions, to use more understandable names
Migrate layers to a naming convention based on `activate`/`deactivate`, while keeping the old methods around, as deprecated aliases.

Fixes #249.
closed 2018-12-10 by obra