keyboardio/Kaleidoscope #250: Layer::on needs to be documented.

obra, 2017-11-23:


gedankenexperimenter, 2017-11-23:
I just added a one-line description for both `on()` and `off()` in #230. This made me realize that I'd like some guidance on how to document functions in source code comments. I'd like to do a whole bunch of that as I work through the code and try to understand it, but I hesitate to do it because I don't know if you want it done in any particular format. Should I be trying to write comments for Doxygen to interpret?

algernon, 2017-11-23:
> Should I be trying to write comments for Doxygen to interpret?

Yes, please!

[Here](https://keyboardio.github.io/Arduino-Boards/) is how the docs look right now, and we are indeed using Doxygen. Mind you, this whole thing is at an early stage at the moment, there are probably a lot of things we haven't figured out yet.

gedankenexperimenter, 2017-11-24:
I will do my best, with the time that I have. Do you have any recommendations for dividing up the work into separate PRs? One huge one per module? One per file? Per function? In case I really get going and do a lot in one day, I don't want to make things inconvenient for the reviewers by creating too many or by creating ones that are bigger than desired.

algernon, 2017-11-24:
As far as I'm concerned, anything works, though one PR per function is a bit too granular. What I would suggest is one big PR, updated with new commits whenever they are ready. Initially, it would be a WIP PR, something for review, but not ready for merging. Put as much or as little in it as you see fit, and keep adding commits. It can be continously reviewed and refined this way.

Once we reach a state where it makes sense to merge it, we do that. If there's anything left to document after that (and this is the likely case), another, similar PR could be made.

gedankenexperimenter, 2017-11-24:
Excellent! I will go ahead and do that, then.

algernon, 2020-01-06:
We do not have `Layer::on` anymore. Not that the new methods are documented, but.. we have #629 for that, which is broader than this issue. I think we should keep that, and close this. Or retitle this, and close that, since this has been here earlier and has more history.

closed 2020-01-07 by obra