keyboardio/Kaleidoscope #261: Setup instructions should specify actually launching the Arduino IDE

benshine, 2017-12-04:
The setup instructions in the README specify _installing_ the Arduino IDE, and then assume that `$HOME/Documents/Arduino` exists. I just went through a from-scratch installation and noticed that `$HOME/Documents/Arduino` is created on the first _launch_ of the Arduino IDE, not on install. 

Suggestion: add a third bullet point to the `Setup the Arduino IDE` section: 
> Start up the Arduino IDE by double-clicking /Applications/Arduino. This is necessary to create `$HOME/Documents/Arduino`. From this point on, you don't need to use the IDE.

algernon, 2017-12-04:
Either that, or have the instructions create it, so one does not need to launch the IDE just to create a dir...

obra, 2020-06-08:
I believe this issue is obsolete. Please feel free to reopen if it's still an issue.

closed 2020-06-08 by obra